silvanmelchior/RPi_Cam_Web_Interface #408: RPI phpindex no longer loads cam control window

danindiana, 2018-03-20:
*Great* Software, easy setup, and runs like a champ and have had no issues until I attempted to condition the RPI as a web server. 

I think it's a port or index / directory issue which should be easy to resolve but not knowledgeable enough to know what next step is. Run update.sh and enter in correct values? I can also post the phpindex browser debug info if that helps. 

Sorry if this issue is reposted elsewhere, yes I did check the issues and wasn't able to locate exact same problem or resolution of issue. I know a lot of people want to use their Pi's as a web cam AND a web server so...

Think something got misconfigured when I performed the following on the Pi:

  822  sudo apt install apache2
  823  sudo chown -R pi:www-data /var/www/html/
  824  sudo chmod -R 770 /var/www/html/
  825  get -O check_apache.html http://127.0.0.1
  826  wget -O check_apache.html http://127.0.0.1
  827  cat ./check_apache.html
  828  sudo apt install php php-mbstring
  829  sudo rm /var/www/html/index.html
  830  echo "<?php phpinfo ();?>" > /var/www/html/index.php
  831  sudo apt install mysql-server php-mysql
  832  sudo mysql --user=root
  833  mysql --user=root --password=******
  834  sudo apt install phpmyadmin
  835  sudo apt install phpmyadmin mysql --user=root --password=******
  836  sudo apt install phpmyadmin --user=root --password=******
  837  sudo apt-get install phpmyadmin php-mbstring php-gettext
  838  sudo dpkg-reconfigure mysql-server-5.5
  839  sudo service mysql stop
  840  dpkg --get-selections | grep mysql-server-
  841  echo dpkg --get-selections | grep mysql-server-
  842  /etc/init.d/mysql stop
  843  sudo apt install phpmyadmin
  844  systemctl start mysql
  845  sudo apt install phpmyadmin
  846  htop
  847  sudo reboot

[image]

[image]

roberttidey, 2018-03-20:
As I understand this you want to use the RPi Web cam and also use Apache to host other web apps.

If so that's fine but you need to do things in the right order and also understand the options on running multiple web apps.

I would recommend starting by installing the RPi_Cam software before doing anything else. This will install Apache2 and php so you do need to separately install these later.

During the RPi Cam install you can choose the web subfolder (under /var/www) for the RPi cam and the port number it will use. These default to html and 80. You can change these after the install just by running install again. You can also choose to install into the web root folder by blanking out the subfolder.

One normally allows more than 1 web app on the single app server either by using different urls or by using different port numbers. So with the first method one uses http://ip/webapp1, http://ip/webapp2 etc. With the second method one uses http://ip:port1, http://ip:port2 . This assumes that there is a default start page index.html or index.php for each app in their own folder.

The way RPi Cam is set up makes it easier to do the first scheme where the different apps all use port 80 but are in different subfolders. This means RPi Cam is accessed by http:/ip/html and a second app would be installed into different subfolder and accessed by http://ip/webappfolder.

One of the apps can be put into the base /var/www folder and would then be accessed by http://ip

When you install a further web app then you must make sure it goes into a different folder and doesn't overwrite any RPi Cam files, and you need to check or edit the apache2 site config files to make sure they are set up to access their own app.

danindiana, 2018-03-21:
@roberttidey Thanks so much for this informative post. Yes, the Cam software is back up and running after a re-install. I figured it would be matter of proper configuration and allocation of correct directories, not knowing enough beyond the basic bootstrap for apache and php install (on the pi). I'm going to reference your above post and see if I can't use that to disambiguate the various installation configs. Thanks so much for your very informative response!

closed 2018-03-21 by danindiana