silvanmelchior/RPi_Cam_Web_Interface #411: Timelapse conversion issues

ramkumarkr, 2018-03-30:
When I try to run the conversion command
gst-launch-1.0  -v multifilesrc location="tl_0000_%04d_%08d_%06d.jpg" caps=\"image/jpeg,framerate=\(fraction\)30/1\" ! jpegdec ! queue ! videoscale ! video/x-raw ! queue ! omxh264enc target-bitrate=15000000 control-rate=variable ! video/x-h264, profile=high ! h264parse ! mp4mux ! filesink location=
, I am seeing this error
WARNING: erroneous pipeline: no element "multifilesrc"

I tried to tweak the location parameter, but no success. Can anyone spot what is missing here?

roberttidey, 2018-03-30:
Are you trying to run that command manually?

I don't think that multisrc  location is going to work with multiple parameters. When the internal conversion is used it gathers all the files together in one temporary folder with simplified names with just a single incrementing number (location=i_%05d.jpg) The program also inserts the temporary folder name.

There have been problems with latest stretch due to some bits of gstream not being updated. I have tried to see what is wrong in the past but will have to need to look again.

If you want to experiment with it manually then temporarily edit the preview.php file. Around line 192 there should be a line starting $cmd =

Just edit out the rm -rf $tmp; from this line. This will stop the temp file getting deleted. You can now try to run the internal conversion method. The files will be left in their temp folder so you can then run the command again manually

ramkumarkr, 2018-04-01:
It didn't run from the interface and then I was trying to manually run to see if I can find errors. I will try to do the changes you have suggested to avoid file removal, but so far nothing got removed as well (strange).