silvanmelchior/RPi_Cam_Web_Interface #412: Lots of freezes since 6.4.21 - not in 6.4.17

konstantinj, 2018-04-03:
Hi,

I've recently updated my docker image from 6.4.17 to 6.4.21. Since then I experience freezes on all of my cameras. 1-2 per day. I'm solving this with automated reboots.

Can I see the changes in the binary since then somewhere? Nothing that you've changes in this repo could influence this I guess.

roberttidey, 2018-04-03:
The raspimjpeg source is at https://github.com/roberttidey/userland. 

6.4.17 was Feb 20 overall version with a raspimjpeg 5.8.07. There has only been a couple of relatively minor code changes since then.

I think the compiler version used may be different on the latest. I'll look into that.

nerdymark, 2018-08-26:
I also get lots of freezes. I manually compiled from source using the build instructions and still get freezes a couple times per day.

nerdymark, 2018-08-28:
I'm using Stretch on a Zero W. I'll do some tests with Jesse and different versions of gcc.

nerdymark, 2018-09-11:
Same results on recent build in Jesse on both a Pi Zero W and a Pi B v2 on current build with builtin raspimjpeg binary. Freezes after left running for a while, an hour or so I'd estimate.