silvanmelchior/RPi_Cam_Web_Interface #415: Lost access for user pi to GUI when execute install

sbeuvain, 2018-04-08:
Hi, 
I lost access for user pi to GUI after reboot when I execute with user pi : 
`RPi_Cam_Web_Interface_Installer.sh install`

I think that chmod modify rights for user pi because when I try with another user, it's ok.

roberttidey, 2018-04-08:
I don't recommend using that install method. Use the ./install.sh as documented in the wiki,

The RPi_Cam_Web_Interface_Installer.sh is left in for historical reasons as it was developed by another contributor.

sbeuvain, 2018-04-08:
I do that you say but when i execute `./install.sh` then after  : 
```
 mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
sh: 1: /var/www/cam/macros/error_hard.sh: Permission denied
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
```

roberttidey, 2018-04-08:
Those would normally indicate a problem accessing the camera like a cable issue. Make sure this is firmly in at both ends and also the little cable on the camera itself is pushed down.

If you have been trying the other install method it may be best to start again with a clean OS.

sbeuvain, 2018-04-08:
I do a clean OS before I used the method install that you preconise.
I think that I have problem with the camera access because when I do `raspistill -o image.jpg`, it's return errors.
Thanks for your help.

roberttidey, 2018-04-08:
OK.

The two most likely physical causes are the cable connections as previously described or the power supply. The camera draws more current and it is important to have both a good power supply and a low resistance USB cable. Many USB cables have a high resistance and cause a volt drop. Short thick ones are best.

I assume you have turned on camera support in raspi-config.

sbeuvain, 2018-04-08:
Yes, I have turned on camera support in raspi-config.
There is no USB cable. This is Raspberry pi3 with Raspberry PI camera V2.1 with a short cable connector ( white, and blue )
I think too that my son, has inversed connection on slot and he breaks the camera.

Again thanks