silvanmelchior/RPi_Cam_Web_Interface #418: UTF-8 coding for user annotations

drtp, 2018-04-13:
Hi,

would it be possible to include UTF-8 coding for the user annotations? I have the issue that for instance the strings "°C" or "Tür" are not displayed correctly. Instead, the user annotation then shows "??C" or "T??r".

Have a nice day,

Thorsten

roberttidey, 2018-04-13:
Unfortunately the coding and rendering of the annotation is outside of my control. It is done in the MMAL camera interface.

You could try raising it as an issue in the general raspberry userland github. I suspect, however that the rendering is optimised for simplicity and speed as it has to be done at frame rate.

drtp, 2018-04-13:
Thank you very much. Well, since it does work without UTF-8 coding so far, it is not really an important issue (in fact it even isn't one). I only have to omit such characters. However, perhaps I will contact the guys from the MMAL camera interface team in the near future.

Best regards,

Thorsten