silvanmelchior/RPi_Cam_Web_Interface #422: motion_event.sh only on Motion Stop

drtp, 2018-04-26:
Hi,

I am currently utilizing a Raspi Zero WH in combination with an IR camera module and RPi_Cam_Web_Interface in order to detect letter box inserts. If anybody is interested, my project is described [here](https://homematic-forum.de/forum/viewtopic.php?f=18&t=43306#p431626), but in German language only. If a motion start or stop is detected, the interface calls the macro file motion_event.sh and sends me a camera snapshot via Telegram and/or Pushover. So far, so good. However, I only want to obtain a snapshot after motion stop and not on motion start. How can I prevent motion_event.sh of being called on motion start? The parameter "im" within the scheduler seems not to have any influence on the Shell script. 

Would it be possible, to call the number of change frames to start and still frames to stop within the Schell script in order to create a respective "if" dependency?

drtp, 2018-04-26:
I will try it with "um 8 -motion_event.sh" under "Motion Start" and "um 8 motion_event.sh" under "Motion Stop" within the Scheduler.

roberttidey, 2018-04-26:
motion_event.sh is called when either a motion start or stop event occurs, but it is passed a parameter which indicates what type of event it is (1=start, 0 = stop). THis is referenced as $1 within the script.

So the script can an if statement to determine what action to take based on whether it is a start or stop event.

roberttidey, 2018-04-26:
Yes. That's the right idea. I am not sure about the if condition though as I think the 0 and 1 are quoted on the command line. If they survice then you may need to use

if [ $1 == "0" ]

drtp, 2018-04-27:
Yesterdey, I tested it and with

`if [ $1 -eq 0 ] then... fi`

it works perfect. Thanks for your support.