silvanmelchior/RPi_Cam_Web_Interface #423: null pointer crash with motion enabled

thanhtranhd, 2018-05-01:
Hi,

I'm setting up a new camera with RaspberryPi Zero W using this RPi_Cam_Web software.  I keep seeing some crashes like this when motion (internal) is enabled. 

[   59.824723] random: crng init done
[79245.734326] Unable to handle kernel NULL pointer dereference at virtual address 00000008
[79245.743888] pgd = d4680000
[79245.746129] [00000008] *pgd=14d14831, *pte=00000000, *ppte=00000000
[79245.749593] Internal error: Oops: 17 [#1] ARM
[79245.752878] Modules linked in: cmac bnep hci_uart btbcm bluetooth brcmfmac brcmutil cfg80211 snd_bcm2835 rfkill snd_pcm snd_timer snd bcm2835_gpiomem uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
[79245.760794] CPU: 0 PID: 482 Comm: VCHIQ completio Not tainted 4.9.59+ #1047
[79245.762921] Hardware name: BCM2835
[79245.765694] task: d4fe28e0 task.stack: d46d4000
[79245.768412] PC is at vchiq_ioctl+0xfd8/0x188c
[79245.770530] LR is at mutex_lock+0x1c/0x48
[79245.772643] pc : [<c04c6364>]    lr : [<c05d32dc>]    psr: 80000013
               sp : d46d5e40  ip : d46d5e28  fp : d46d5efc
[79245.776997] r10: 00000000  r9 : 0069c791  r8 : 00000000
[79245.779178] r7 : d69e6100  r6 : d69e5000  r5 : c090b170  r4 : d46d4000
[79245.782720] r3 : 00000000  r2 : 00000000  r1 : 0069c791  r0 : 00000000
[79245.785656] Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[79245.788194] Control: 00c5387d  Table: 14680008  DAC: 00000055
[79245.790577] Process VCHIQ completio (pid: 482, stack limit = 0xd46d4188)
[79245.792978] Stack: (0xd46d5e40 to 0xd46d6000)
[79245.795369] 5e40: 00000000 00000000 00000001 00000000 d46d5e7c d46d5e60 c0047d38 d69e5824
[79245.801478] 5e60: b6d22d84 c014c407 d7580194 00001100 b6d22d84 c014c407 c007c8ec 00000008
[79245.806434] 5e80: ffffffff 00000000 00000000 b6f3b000 00000008 b6d22db8 00001000 00000008
[79245.811569] 5ea0: b6d22d98 b6f3b3c0 00000001 d46d4000 d46d5f4c d46d5ec0 c007e214 c007c7dc
[79245.818713] 5ec0: d46d5eec d46d5ed0 c004bc68 c00401a0 d4c23d40 b6d22d84 d73822f8 d46b2000
[79245.824443] 5ee0: 00000004 00000004 d46d4000 00000000 d46d5f7c d46d5f00 c01575f0 c04c5398
[79245.830112] 5f00: d46d5f24 d4fe28e0 00000000 d4c23d40 00004812 c08b17f0 d46d5f74 d46d5f28
[79245.836874] 5f20: c05d1910 00000000 00000001 00000081 b6f3b3c0 c016211c d7080840 b6405550
[79245.842599] 5f40: b6d22d84 00000001 c014c407 00000004 d46d5f6c d46b2001 b6d22d84 d46b2000
[79245.849862] 5f60: c014c407 00000004 d46d4000 00000000 d46d5fa4 d46d5f80 c0157d6c c0157560
[79245.856684] 5f80: 00000008 b6405550 b6ee72ec 00000001 00000036 c000ffc4 00000000 d46d5fa8
[79245.862890] 5fa0: c000fe40 c0157d34 b6405550 b6ee72ec 00000004 c014c407 b6d22d84 00000008
[79245.870301] 5fc0: b6405550 b6ee72ec 00000001 00000036 b6ee7308 b6d22d84 b6ee72b4 b6ed6f3c
[79245.876610] 5fe0: b6ee7240 b6d22d54 b6ed4010 b6ded8ac 20000010 00000004 5767f57c 2f379dcf
[79245.884815] [<c04c6364>] (vchiq_ioctl) from [<c01575f0>] (do_vfs_ioctl+0x9c/0x7d4)
[79245.891459] [<c01575f0>] (do_vfs_ioctl) from [<c0157d6c>] (SyS_ioctl+0x44/0x6c)
[79245.898283] [<c0157d6c>] (SyS_ioctl) from [<c000fe40>] (ret_fast_syscall+0x0/0x1c)
[79245.905349] Code: e0867003 e50b3090 e597300c e597a008 (e5932008) 
[79245.936746] ---[ end trace ce62fcdaf191c273 ]---

Once it crashes, the image on the web browser freezes. I would need to reboot the Pi Zero W to restart. 

I'm using  Raspian 2017-11-29-raspbian-stretch-lite.
It also happens on 2018-04-18-raspbian-stretch-lite.img

Both are fresh install and has nothing else running. Both boot into text console. 

If I don't enable motion (Internal), it would run for days without problem.

Any idea / suggestion on where I should look at to debug this?

Thanks
-Thanh

thanhtranhd, 2018-05-02:
Hi Rob
Thanks for your help.
I see this at the end of the /boot/config.txt file

# NOOBS Auto-generated Settings:^M
hdmi_force_hotplug=1^M
start_x=1
gpu_mem=128

The crash seems to happen at random places.  
1.
[ 1637.909066] Unable to handle kernel NULL pointer dereference at virtual address 00000070
[ 1637.915969] pgd = c0004000
[ 1637.920743] [00000070] *pgd=00000000
[ 1637.925871] Internal error: Oops: 17 [#1] ARM
[ 1637.929694] Modules linked in: cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic evdev joydev hid_logitech_hidpp hid_logitech_dj brcmfmac brcmutil snd_bcm2835(C) snd_pcm snd_timer cfg80211 snd rfkill uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
[ 1637.941180] CPU: 0 PID: 710 Comm: kworker/0:2 Tainted: G         C      4.14.34+ #1110
[ 1637.949368] Hardware name: BCM2835
[ 1637.956712] Workqueue: mm_percpu_wq drain_local_pages_wq
[ 1637.961005] task: d50fe120 task.stack: d50b8000
[ 1637.964641] PC is at free_pcppages_bulk+0x274/0x498

2.
Message from syslogd@picam2 at May  1 01:58:43 ...
 kernel:[ 1746.319650] Internal error: Oops - undefined instruction: 0 [#1] ARM

Message from syslogd@picam2 at May  1 01:58:43 ...
 kernel:[ 1746.364325] Process VCHIQ-0 (pid: 60, stack limit = 0xd7386188)

3.
Message from syslogd@picam2 at May  1 02:35:29 ...
 kernel:[ 1689.010362] Internal error: Oops: 17 [#1] ARM

Message from syslogd@picam2 at May  1 02:35:29 ...
 kernel:[ 1689.057757] Process MP4Box (pid: 676, stack limit = 0xd4d70188)

Not sure if they ring any bell?

I'll try the full install to see if it behaves differently.

Thanks

-Thanh

thanhtranhd, 2018-05-04:
I tried the full NOOB image and the problem is also there. 
It looks like this could be isolated to a board and a camera module that were running the RpiCam software. 
Once I exchange the Raspi Zero board the problem disappeared.  However, when using that Raspi ZeroW board with the same micro SD card (that was having issue) with a different camera module then the problem does not show up either.  I've tried changing power supply, USB cable.  Maybe I have cabling issue on that camera. Can't try it yet as I don't want to disturb that working set up now.  When I have another of those camera module, I'll try again.  Thanks for your help

closed 2018-05-04 by thanhtranhd

anandbaskaran, 2019-04-08:
I am getting the same error in `Raspi Zero W`. I too think it is because of the connectivity issue.

roberttidey, 2019-04-08:
Is the software working OK without motion detection off?

Are you using internal or external motion detection? Internal is much better.

Kernel crashes can also be caused by power supply problems and loose connections to the camera. As the camera draws extra current you need a good power supply and also importantly a good short USB power lead. Many USB cables have a high resistance and cause voltage drops. The camera ribbon should be checked for being well inserted both ends and the little ribbon on the camera module itself should be pressed down as it can sometimes become loose.

thanhtranhd, 2019-04-09:
In my case, I narrowed down to a Raspi zero W board itself that caused the exception.  The same camera module and its flex cable and the same micro SD card work fine on a RaspiZeroW board. Switching the the RaspiZeroW board with the problematic one the exception showed up.

roberttidey, 2019-04-09:
OK. It does sound like a problem board.

Check the camera socket on the board. They are quite fragile and if cracked then it may not apply enough pressure to make good contact. If holding it down works then one can sometimes repair it with a bit of tape.

The other thing could be a dodgy usb connector. Try connecting power into the other if it is free, or if you can wire %V,0V direct to the GPIO connector that would bypass any problem.