silvanmelchior/RPi_Cam_Web_Interface #426: Cannot execute motion_event.sh

ffcruz85, 2018-05-21:
I am having some problems running the motion_event macro. I have followed the docs (https://elinux.org/RPi-Cam-Web-Interface#Job_Macros) and this tutorial (https://quavoce.wordpress.com/2017/06/04/pushover-alerts-for-motion-detection-with-rpi-web-cam-interface/). The logs say that the macro is being executed, but nothing results from it.  I have tested running the bash script on the command line and it works fine, but not when rpi_cam_web_interface tries to run it. I am posting a snippet from the log, and then also my steps to get there.

Any ideas are appreciated! Thanks!

LOG SNIPPET:
```
{2018/05/21 06:17:23} Set up internal detect width=121 height=68
[2018/05/21 06:17:38] Logged in user::
[2018/05/21 06:17:38] UserLevel 6
{2018/05/21 06:17:41} Executing macro /var/www/html/macros/motion_event.sh "1" 
{2018/05/21 06:17:41} send smd 1
[2018/05/21 06:17:41] Start capture requested from Pipe
[2018/05/21 06:17:41] Send ca 1
{2018/05/21 06:17:41} Capturing started
{2018/05/21 06:17:49} Executing macro /var/www/html/macros/motion_event.sh "0" 
{2018/05/21 06:17:49} send smd 0
[2018/05/21 06:17:49] Stop capture requested
[2018/05/21 06:17:49] Send ca 0
{2018/05/21 06:17:49} Capturing stopped
{2018/05/21 06:17:49} Add /var/www/html/media/vi_0002_20180521_061741.mp4 to Box Queue at pos 15
{2018/05/21 06:17:50} Start boxing /var/www/html/media/vi_0002_20180521_061741.h264 to /var/www/html/media/vi_0002_20180521_061741.mp4 Queue pos 15
{2018/05/21 06:17:51} Finished boxing /var/www/html/media/vi_0002_20180521_061741.mp4 from Box Queue at pos 15
{2018/05/21 06:17:51} Removed item from Box 
```

MY STEPS TO REPRODUCE:

`sudo nano /var/www/html/macros/motion_event.sh`

added the following code to the script:
```
#!/bin/bash
echo "some data for the file" >> testmacro.txt
```
please note that what I actually want to do is use aplay to play a sound and then send some pipe commands to change the look of the video - it is for an art performance, just keeping it simple for test purposes. Regardless, even this simple doesn't work, although the script works when run from the terminal as pi user

```
sudo chmod -x /var/www/html/macros/motion_event.sh
sudo chown www-data:www-data /var/www/html/macros/motion_event.sh
sudo chmod 764 /var/www/html/macros/motion_event.sh
```

roberttidey, 2018-05-21:
The log says the macro executes, so it is likely that the script is not doing what you think it should.

I would put the full path in the macro so you know exactly where the file goes. 

echo "some data for the file" >> /var/www/html/testmacro.txt

I just did exactly that and got the file as expected when motion was triggered.

Note that when you get this working then any commands you use in the macro must be permitted for user www-data. If not then you need to give that user permission to do so.

ffcruz85, 2018-05-21:
Thank you! I am trying that on a fresh install right now - a quick follow up question: I need the script to run aplay so I can play a sound through the headphone jack. I have managed to do that through the command line, but I assume that to run that in the script, www-data would need some permission. How can I give it permission to run aplay for a wav file? and should I place that wav file in the macros folder? do I need to give permissions to the file itself?

Thank you so much for your help!!! I spent a very frustrating full day on this!

ffcruz85, 2018-05-21:
YES! I can't believe it was that simple. I managed to create the file with echo.

So, I am proceeding with the next thing, which is to play the sound. I did two things:

1) Changed the script to this:
```
#!/bin/bash
aplay /var/www/html/macros/beep.wav
echo "played sound" >> /var/www/html/macros/testmacro.txt
```
2) Gave ownership of beep.wav to www-data
`sudo chown www-data:www-data /var/www/html/macros/beep.wav`

3) Still didn't work, so also tried this:
`sudo chmod 764 /var/www/html/macros/beep.wav`

Any ideas? it seems maybe I need to give www-data the ability to run aplay, but not sure how to do it... visudo something?

roberttidey, 2018-05-21:
Not sure about the crash which sounds strange.

I think you may be able to allow user www-data to play sounds by adding the user to the audio group

`usermod -a -G audio www-data`

roberttidey, 2018-05-21:
I think a possible reason for the crash is that the motion script is not set up to run asynchronously so if it stalls then it will stop the main program.

One can fix that by changing the config file /etc/raspimjpeg so the motion_event is defined like

`motion_event &motion_event.sh`

I will change the install to make that the default.

ffcruz85, 2018-05-21:
I think the crash might have been due to other changes I had attempted -  I think it was I was trying to play the sound file while it was located in /home/pi/.
 Adding to the audio group did not work,, but I really appreciate your help and you have gotten me to where I am running the script and sending pipe commands to change the images and all that is working!

I also tried separating the aplay command into a different a script, that did not work.

I am going to investigate other solutions like adding www-data to other groups and maybe also giving it permissions - maybe specific one or even full root (i know that's unsafe, but it's in a closed, unconnected network and being used for a few hours for an art performance event, where a video projection is gamefied so that when users move out of the motion mask all of these things happen)

In any case - please let me know if you have any other ideas I can try. Robert - I really appreciate your help!

roberttidey, 2018-05-21:
If the group method doesn't work then you can give www-data specific permission to use aplay via the sudoers facility. This is already used to allow www-data to use the shutdown command.

Basically you need to have a line in the /etc/sudoers.d/RPI_Cam_Web_Interface set up which permits www-data to run the /usr/bin/aplay command

ffcruz85, 2018-05-21:
SUCCESS! That worked. Thanks for all the help Robert, I really appreciate you working through this with me... now to all the new things I will play/tinker with using this! Certainly an upgrade to the RPI-Cam-Web-Interface running my front door camera :)

closed 2018-05-21 by ffcruz85