silvanmelchior/RPi_Cam_Web_Interface #427: Live preview through HDMI

ffcruz85, 2018-05-21:
So I have a setup to play some sounds and change video appearance through a motion_event macro. It works great based on the instructions given in issue #426 

This is for an interactive projection on a large screen. The preview on the web GUI, however, is a bit low resolution and has some lag. I have tried running it on a browser with 127.0.0.1 on the pixel desktop but it is still not ideal. I was wondering what's the best way to get the live preview directly through the HDMI output. I was thinking of two possible solutions described below, which I tried, with varying results described below. ANy other ideas on how I can get the best preview? Preferably raw straight out of the HDMI? It would be ideal to not do it over wi-fi or the web gui.

1) SOLUTION 1 -> Attempted to use raspivid or python camera preview, but since rpi-cam-web-interface is using the camera, it throws an error saying the resource is not available. Regardless, that would not show the visual effects being created through the motion_event script. Is there some way of getting a "raw" output of the live preview through the HDMI? like essentially going straight to the source of what the web GUI is receiving?

2) SOLUTION 2 -> Ran a browser on the Pixel desktop on 127.0.0.1 in hopes of reducing lag. Lag was minimal, but resolution not ideal. Used browser in full screen on min.php page. Also, the screen ration was different, resulting in a scrollbar. I think Solution 1 would be ideal, but if not possible, I was hoping to learn how to get the best possible quality preview on the browser, and hopefully, with the same ration as the screen of the pixel desktop. I have the video on rpi-cam-web-interface set at 1920x1080, with the desktop resolution being the same.

Any ideas would be greatly appreciated!

roberttidey, 2018-05-22:
You can control the resolution of the preview by adjusting the preview quality settings under camera settings. These default to values which give reasonable quality while keeping network bandwidth reasonable. In particular the width setting (default 512) controls the resolution. The height is computed automatically to keep the aspect ration the same as the video. So you can increase that value up to the width of your native video resolution. The preview quality controls the amount of compression used. You could increase that to reduce any compression artifacts.

I would stick with solution 2 as you do need something to render the stream to video output which is what the browser is doing. You could potentially use other apps than a browser to render the stream e.g. vlc but you may need to compile that to get a hardware accelerated version

ffcruz85, 2018-05-22:
Thank you. I am slowly tweaking those settings to get the best compromise between quality/performance. Perhaps this is an academic question at this point, but is this what the jpglink question in the installation is for? I don't see a lot of documentation about it out there. I tried to created a stream in VLC with the URL to the cam.jpg file but could only get a static image - didn't try very hard, was just experimenting.

roberttidey, 2018-05-22:
The jpglink is unrelated to streaming. It just provides a link to the cam.jpg image from the web folder. This isn't needed for the basic operation of the software but is provided for some external apps that want direct acces to cam.jpg rather than using a php URL

For vlc you would need to set its stream to http:/ip/html/cam_pic_new.php?pDelay=40000

where pDelay can be left out but is used to control the frame rate of the stream in microseconds. 40000 gives a frame rate of 25Hz

ffcruz85, 2018-05-22:
Thank you Robert! That makes sense now, I was complicating things. While we are on performance, one last question. In terms of improving performance of the stream and rendering - is it better to favor the CPU or the GPU in the memory split? 
Concerned about bandwidth, I originally installed full Raspbian with desktop (which I usually don't) in order to stream on chromium locally and project there. Then I found that it did not have much advantage over streaming it because of the clunky performance of the desktop, so now I am projecting from an outside PC, and just installing Raspbian Lite so it is a little less bloated. Anyway, this is a long way to ask -> where does the processing happen? if I am not using the desktop, and just installing Raspbian lite to stream over Wifi, should I favor the CPU or the GPU in the memory split?

roberttidey, 2018-05-22:
There is the main underlying raspimjpeg process which interfaces to the camera and extracts multiple streams of data out.  Most of this is employing the GPU for encoding, scaling, preview and raw motion data which is why the CPU usage is pretty low. Providing the GPU has sufficient memory for these tasks (~128MB) there is no significant performance benefit in allocating more.

The CPU is used for running the apache web server and for doing some of the motion detection analysis but these are relatively lightweight tasks particularly using the internal motion detection which uses the GPU to provide vector motion data. Typically I see < 5% overall CPU usage running the raspimjpeg and web server. I don't know how much CPU the browser would take if running locally n the raspberry. If you run the top command from a terminal that will give you a good idea of cpu usage across the various tasks.

There is a paypal donate link referenced on the wiki 

 https://www.paypal.com/cgi-bin/webscr?cmd=_s-xclick&hosted_button_id=FWX7UL9NSNM6L

xxLeoxx93, 2021-01-30:
Hello,
I've pretty much the same usecase (running on a Pi Zero with an 7inch screen attached). However when I use http:/ip/html/cam_pic_new.php?pDelay=40000 in VLC I get like 1 frame a sec with a huge delay and the log shows "blending rgba to mmal failed". is there any chance on how to improve that? I want to use it as a video doorbell with the screen inside as well as remote access.

Thanks!

roberttidey, 2021-01-30:
Few questions.

What sort of screen and screen driver are you using? I have used a local screen hdmi attached and got normal refresh rates.

Have you looked at processor usage (e.g. using top command)? Normally on a zero usage should be below 10%

Have you tried testing it from a remote browser without using the local display? This will allow separating out server issues from client issues.

pDelay does work.

I am not sure I understand the comment on display start up time. Aren't you keeping a live client going all the time?

xxLeoxx93, 2021-01-30:
Hey there,
Thanks for the quick response!
Everything works flawlessly when I open the webinterface on any other device then the pi zero where the camera is attached to. Wit the startup time I meant the time that the pi zero needs to load up a Webbrowser. 
the Display is connected via hdmi and is a 60hz panel with a 1024x600 resolution.
CPU usage is, without doing anything on the pi zero besides running the Webserver, at around 10%.
I‘m on default raspberry pi os with no special driver installed. Could it be that the pi zero is just not powerful enough to render the mpeg stream?

Regarding the parameter: pDelay=40000 means 25fps so pDelay=1000000 should be 1fps right?

Regards Leo

roberttidey, 2021-01-31:
Pretty much all of my usage is with separate remote browser clients, but I did test some time ago with a client on the server pi. I'll set up to re-run the test.

Yes. pDelay is microseconds between frames so 1000000 is 1 second or 1 fps.

Depending on the browser you may be able to turn on development tools which can show what is going on. It is worth turning off the mjpeg streaming option for this as you can then see each individual frame in the network view.