silvanmelchior/RPi_Cam_Web_Interface #432: [question] How does RPi Cam Web Interface work?

PortStone, 2018-06-10:
Hello,

I didn't really find something on the internet that explains silvanmelchior RPi Cam Web Interface. I want to know how does it work. I just want to know how the lag free live stream works. At the moment, I only know that it makes a picture and a PHP script shows the latest image all the time. This can not be all or?

Kind Regards 
PortStone

roberttidey, 2018-06-10:
RPi Cam web interface is made up of two parts; a raspimjpeg process and a web server.

The raspimjpeg code is at the repository in the wiki. It provides a number of functions by using the MMAL interface to the camera system. It can simultaneously open up several connections to the camera data to provide image and video recording, the production of preview data, and the detection of motiion based on vector data from the camera. It makes extensive use of the GPU so the CPU usage is very light.

Normally whilst it is idle it provides a preview by getting raw video data at video frame rate, rescaling it to the desired resolution and compressing it to jpeg format.

The web side can either feed the preview images one at a time to the browser, or can combine them into a mjpeg stream to the browser.

Independently of the preview stream, the raspimjpeg process accepts commands via its pipe interface. These can include video start / stop and image capture. When video capture is started then a separate channel to the camera / GPU is opened up to record h264 encoded video at whatever resolution / frame rate has been selected. This happens in parallel with the continuing preview feed. Likewise with image capture. When a video recording ends the raspimjpeg process normally runs a MP4Box command asynchronously on the h264 video to format it as a mp4 file which browsers can display.

The web side runs under php normally on Apache to display the preview and give access to all the raspimjpeg commands for capture and camera settings. It also runs a scheduling process which can automate various functions and give flexibility to what happens when motion detection occurs if that is used.

The best way to understand in further detail what is going on is to look at the raspimjpeg code base.

PortStone, 2018-06-11:
Thank you. Can be closed.

closed 2018-06-11 by roberttidey