silvanmelchior/RPi_Cam_Web_Interface #433: [question] change dimension size of live preview

PortStone, 2018-06-16:
Hello,

if I right click an image from the live preview and downloading it, it has always the size of 512x288. How can I change this to 1920x1080? I use your project for a school project and just need the live preview. (Remote car control)

Kind Regards
PortStone

PortStone, 2018-06-16:
Oh and it is really smooth but sometimes it stocks. It is installed on a fresh installed Raspbian stretch with GUI on the following Hardware:

Raspberry Pi 3 Model B V1.2
Raspberry Pi Camera Rev 1.3

with Apache-Webserver and PHP 7.0

roberttidey, 2018-06-17:
The resolution of the preview is controlled by the preview settings under camera settings.

You set the width and the height is calculated from the aspect ration of the video.

Stumble can be caused by networking issues. It is good to use mjpeg stream mode (under system settings) to reduce handshaking, Are you using wifi or wired? Wifi will suffer more. You can reduce the amount of network traffic by slowing down the preview rate a bit (divider 2 gives 12.5 fps) and you can also increase the compression (lower quality) a bit.

PortStone, 2018-06-17:
Okay I switched to mjpeg stream but I still don't get it how I can change the resolution. 
[image]

Which settings do you mean?

roberttidey, 2018-06-17:
Preview has 3 settings; quality (qu), Width (WI) and Divider Di

Width (default 512) sets width of preview

Quality sets compression. Lower numbers less quality, higher compression

Divider lowers update rate of preview by the divider factor. 1 = same as video (25fps), 2 is half rate (12.5fps), 3 = 8.33 fps and so on.