silvanmelchior/RPi_Cam_Web_Interface #434: Custom file html

MastroGeppetto14, 2018-06-24:
Hi,

i would like customize your web page, but i don't find your file html. i wanted disable the buttons, can you help me?

roberttidey, 2018-06-24:
The main web page is index.php. This generates and feeds back the html dynamically to the browser.

This odes already support three levels of interface based on user login. Minimum has no buttons at all (preview only), medium has main control and capture buttons but no access to settings, full has access to everything.

The wiki has some details on how to set this up.

If you want to do other customisation and can't see how to do it in index.php then ask further questions.

MastroGeppetto14, 2018-06-25:
Thx mate i resolved ;)

closed 2018-06-25 by MastroGeppetto14