silvanmelchior/RPi_Cam_Web_Interface #435: Notice: Undefined variable: remote_user in /var/www/index.php on line 265

Jacotheron, 2018-07-03:
I am getting this error in my logs (only this RPi_Cam_Web_Interface is installed there). Having turned the Display Errors on (obviously not in production environment), I see this error displayed just above the preview video.

I looked at the source code, and the relevant block is (right before closing the start of the PHP, and starting the HTML output):
```
if(stripos($serverSoftware, 'apache') !== false) {
	   $user = apache_getenv("REMOTE_USER"); 
} else if(stripos($serverSoftware, 'nginx') !== false) {
	   try {
		$user = $remote_user; //<--- Offending line
	   } catch  (Exception $e) {
		$user = '';
	   }
} else {
	   $user = '';
}
```

I also did a quick look through the code but do not find anything that sets the "$remote_user" variable. Thinking that it may be user based, I created a new basic_auth user (since I did not enable it on install), and still the remote_user variable is not populated.

I am using Nginx for the server, as it handles proxy features on another port.

While I can disable the error display and log, I don't usually like doing so, as notices becomes errors and errors can become fatal errors.

roberttidey, 2018-07-03:
$remote_user is supposed to be a nginx embedded variable when authentication is used. http://nginx.org/en/docs/http/ngx_http_core_module.html

This is also why it is in a try exception clause to catch the exception if not present.

I don't normally use nginx myself so I am not an expert on its config, but I thought I tried out the $remote_user some time back. I'll have to check whether the syntax to access it from php is correct.

You would still get the caught exception without a user. It may be possible to config nginx logging to ignore the caught exception. Maybe catch_workers_output

roberttidey, 2018-07-03:
Maybe it should be
$user = $_SERVER['REMOTE_USER'];

If you have a running nginx system could you try that?

Jacotheron, 2018-07-03:
That helped, but in my own research I also found that by default, it seems, nginx does not provide the variable. So in addition to this change, I had to add the following to either the server configuration (inside the /etc/nginx/sites-available, inside the location for PHP files) or the /etc/nginx/fastcgi_params file:
`fastcgi_param REMOTE_USER $remote_user;`

After this change, Nginx needs to be restarted, to take effect, and then the PHP can find it using the `$_SERVER['REMOTE_USER']` as mentioned above.

closed 2018-07-03 by Jacotheron

roberttidey, 2018-07-03:
Thanks. I have changed it to use the $_SERVER['REMOTE_USER'] and the nginx sites-available has the extra parameter added.