silvanmelchior/RPi_Cam_Web_Interface #436: htpasswd corruption after install/update: auth+nginx+stretch+PiZeroW

johnprather, 2018-07-21:
I have reliably reproduced this with every attempt at install.sh and update.sh on two different devices.

Using:

* Raspberry Pi Zero W
* Raspbian Stretch (9.4)
* Nginx (1.10.3-1 installed by install.sh)

Installation choices:

* Server: nginx
* Username/Password: non-blank

Installation completes successfully, however, the web service is not yet ready to run, as it serves up a 500 error.

The first thing I notice in nginx error log is this:

```
2018/07/20 23:41:58 [error] 1807#1807: *3 rewrite or internal redirection cycle while internally redirecting to "/index.html",
```

This error *can* be fixed by updating `/etc/nginx/sites-available/rpicam` and changing `try_files $uri $uri/ /index.html;` to `try_files $uri $uri/ =404;`.  This, however, appears to be a problem which does not *need* to be solved, as solving the *next* error seems to either resolve or workaround this error.   

The blocking error in the nginx log which *must* be addressed to get the web service up in this environment is:

```
2018/07/20 23:42:39 [crit] 2438#2438: *1 crypt_r() failed (22: Invalid argument),
```

Seems this is caused by corrupt/incompatible data in a htpasswd file.  In every install/update, I needed to perform the following to cease the 500-server-error behavior of the web interface (where `admin` is the username I provided when running install/update script):

```
sudo htpasswd /usr/local/.htpasswd admin
```

Upon providing and verifying the password, the site becomes available and both errors mentioned above cease appearing in the nginx error log.

Other than needing to recreate the htpasswd entry, the RPi Camera Web Interface installation script on Pi Zero W running Raspbian Stretch has been a great success!

johnprather, 2018-07-21:
I meant to include that this experience is based on https://github.com/silvanmelchior/RPi_Cam_Web_Interface/commit/fc4d44fbed4ee31702b5aec5268e8fc1b3b4233d.

roberttidey, 2018-07-21:
Thanks for the detailed report.

I normally use Apache and the passwd is OK on that.

Both nginx and Apache use

sudo htpasswd -b -B -c /usr/local/.htpasswd $user $webpasswd

to set up the .htpasswd file.

Might be worth seeing if that command fixes it outside of the installer where $username and $webpasswd are the real names.

I'll set up a nginx install to investigate further.

johnprather, 2018-07-21:
I can reproduce the problem manually with:

```
sudo htpasswd -b -B -c /usr/local/.htpasswd test t3st
```

It prompts for credentials, then delivers the 500.

This, however, works fine:

```
sudo htpasswd -b -c /usr/local/.htpasswd test t3st
```

I guess this nginx build doesn't like bcrypt.

roberttidey, 2018-07-21:
Thanks for that. I have removed that switch for nginx installs