silvanmelchior/RPi_Cam_Web_Interface #439: Multiple Vulnerabilities

reigningshells, 2018-08-12:
I don't want to post too much publicly, as I know a number of people who use this interface and some who have it forward facing.  There are two critical vulnerabilities I'd like to disclose.  Please reach out on Twitter @reigningshells or via email, RPi_Cam_Bugs [at] reigningshells.com.

roberttidey, 2018-08-13:
Thanks. I have received the further details on this now.

I was planning to do some other minor updates this week(2018-08-17) so will see if I can incorporate these suggestions.

roberttidey, 2018-08-21:
Latest version has some changes around these vulnerabilities

reigningshells, 2018-08-23:
I sent a follow up email, just one more slight fix is necessary

roberttidey, 2018-08-23:
I have added some extra checks which should avoid that.

I also discovered I broke some buttons when I removed the convert command dialog box so I fixed that.

reigningshells, 2018-08-24:
Sorry, had some time to dig in and kind of flooded your inbox this evening, but I think a few more calls to checkMediaPath will solve all the issues reported tonight.

roberttidey, 2018-08-24:
Thanks for the extra info on this. I have added the calls in to latest version.

roberttidey, 2018-08-25:
6.4.29 has the exists call in the check function

reigningshells, 2018-08-30:
Looks good, thanks for all your hard work!

closed 2018-08-30 by reigningshells