silvanmelchior/RPi_Cam_Web_Interface #440: Zip files corrupt?

StudioBeneden, 2018-08-28:
Hi! thanks for this well built interface!

I've had RPi Cam working time lapses for weeks on a Rpi 3, and am now trying it out on a Zero-W. Everything works just fine, but for one thing:

When I download a time lapse with the 'get Zip' function on the files page, the file downloads as usual, but after download, the file can't be unarchived by Archive Utility (Mac osX El Capitan). (Unable to expand...(Error 2 - No such file or directory). 

I am able to expand it (with errors) with 'the Unarchiver', but then only half the files are there. When I check "/var/www/media" Alle files are there, downloading over SMB works fine. Looks like the zip process is interrupted somewhere during download. 

I've tried different browsers, and did a fresh install on another SD. Even swapped out the Zero-W

Anyone any insight on what might be happening here?

thanks!

StudioBeneden, 2018-08-28:
Did some more testing, it appears that zip files containing over 500 jpgs fail, everything under 500 files seems to download ok.. When I unpack and check the corrupt archives, there is no consistent number of files in them on different downloads. 

my ulimit 'maxflies' is on 65536. Is there another php/system limitation somewhere?

roberttidey, 2018-08-28:
I'll look into that.  I think I did a test with 2000 files once but that was a long time ago.

roberttidey, 2018-08-28:
I have just done a test downloading a zip with 900 time lapse images init (about 300MByte).

That worked. I could open the archive and could extract images and a quick test showed there were OK. So I don't think there is anything going wrong logically with number of files.

My test was over a wifi connection but was reasonably fast (150 seconds).

It could be that if your zip size is bigger in size or your transfer rate is slower then the browser may be terminating the download due to timeouts rather than the number of files in the zip.

Might be worth opening developer tools in the browser before starting the download. You may get a clearer picture of any errors that way.

StudioBeneden, 2018-08-28:
Thank you for diving into this!

The chrome Developer console only puts out a warning "Resource interpreted as Document but transferred with MIME type application/zip: "http://camera.local/preview.php" at the beginning of the download. No errors. Is there a more detailed way/tool to check what's going on in those transfers?

Meanwhile, I'm pretty sure (like you said too) the problem isn't related to the number (500) of files. repeated testing with a 450 an 503 file TL shows that both sometimes fail, and sometimes don't, on chrome downloads sometimes stop with "failed - Network error", no errors in the developer console however.

The download over my office wifi is reasonably fast (ca. 2 - 2.5Mb/s). And normally pretty stable too. I'll try to set up a dedicated wifi or set up the Zero-W as an access point to rule out my office wifi.

The Zero-W is connected to a PiJuice HAT. I will test without the HAT, just to rule out that the PiJuice is messing with the wifi signal.

StudioBeneden, 2018-09-11:
So, I finally had he chance to do some more testing.

I've tried 2 different Zero-W's from different suppliers, 3 different SD cards, with fresh installs. Running the Zero over a dedicated Wifi, and running the Zero as an AP. Powered from a power adapter and via the PiJuice.

Downloading kicks off with 2 -2.4Mb/s and then fails without warning. Presenting an incomplete Zip file in my Downloads folder. Downloads above 250Mb Fail 90% of the time. For 500+ Mb files downloading stops around 300Mb. And fails 99%.. I've tested on Windows and on Mac, on Edge, Safari, Chrome, and FireFox... 

Today I took out the card with the latest install and a video-recording of 1,5Gb, popped it into my Raspberry Pi 3 Model B V1.2.... downloaded the full 1,5 without fail with 2.4Mb/s throughout

Except for price, availability and size circumstances there's not a big issue to create the camera's I'm making around the Pi 3. But it's still bugging me why the Zero fails at this task...

jtewright, 2020-08-02:
Have had the same issue with the web interface get zip option.

I was able to copy out all the images using scp. While **not** ssh'd into the RPi:

`scp -r pi@raspberrypi:/var/www/html/media/*.jpg <location on machine>`

It'll SSH authenticate you and then just start copying.

roberttidey, 2020-08-02:
Is that over wifi? I have tried this over my wifi and it is OK but some users with potentially intermittent wifi can suffer from problems with large downloads. Using scp as you have done is transferring smaller files one at a time.

As suggested in an earlier post it is worth opening the network tab in developer tools in the browser to see in more detail what is going on.