silvanmelchior/RPi_Cam_Web_Interface #441: Missing part in the recorded video with internal motion

monanto, 2018-08-30:


roberttidey, 2018-08-30:
That is not normal.I never get any missing video and I use internal motion all the time.

Few things will help investigate what is going on.

1 Are you using motion_event.sh, and if so what is it doing?

2. What model Pi is this on?

3. If you manually start and stop recordings of 15 seconds is the recording OK?

4. Are you using video buffering (camera settings)?

monanto, 2018-08-31:
Yes
1) Are you using motion_event.sh, and if so what is it doing?
sudo cat /var/www/html/macros/motion_event.sh 
#!/bin/bash

echo "$@"|grep -a '0'
if [[ $? -eq 1 ]];then
#python /var/www/html/macros/udp.py
cp /dev/shm/mjpeg/cam.jpg /var/www/html/cam.jpg
curl 'http://192.168.1.8:5000/motion?motion=2' &
#curl -s  --form-string "token=....."  --form-string "user=......."  --form-string "Motion Alert Camera"  -F "attachment=@/var/www/html/cam.jpg"  https://api.pushover.net/1/messages.json
curl -s  \-F "token=apb26m5vcifusqfsfsdpmye6f6fqqy"  \-F "user=utmsq7toeni9jux4dq3oh2serx64vn"  \-F "message=Motion Alert Camera"  \-F "attachment=@/var/www/html/cam.jpg"  https://api.pushover.net/1/messages.json &

fi

monanto, 2018-08-31:
I think i found the problem
I changed motion_event.sh to

#!/bin/bash
echo "$@"|grep -a '0'
if [[ $? -eq 1 ]];then
#python /var/www/html/macros/udp.py
cp /dev/shm/mjpeg/cam.jpg /var/www/html/cam.jpg
curl 'http://192.168.1.8:5000/motion?motion=2' &
#curl -s  --form-string "token=....."  --form-string "user=......."  --form-string "Motion Alert Camera"  -F "attachment=@/var/www/html/cam.jpg"  https://api.pushover.net/1/messages.json
curl -s  \-F "token=......."  \-F "user=..............."  \-F "message=Motion Alert Camera"  \-F "attachment=@/var/www/html/cam.jpg"  https://api.pushover.net/1/messages.json &
fi

I think the time it takes to process motion_event.sh That amount of time clip gets missing

Thanks for your help

roberttidey, 2018-08-31:
OK. That's good.

The macro scripts can be launched synchronously or asynchronously. Synchronous means that nothing else will happen on the camera side till macro completes. Asynchronous means it is launched as a background task and returns immediately to the camera software. 

Synchronous is sometimes useful where you want to do some short action that might influence the camera operation.

The difference is determined by the set up in /etc/raspimjpeg  Each macro can have a & in front of the filename. Where this is present then the macro is launched asynchronously. The default for the motion_event.sh was synchronous.

So I suspect you could have used your original version if it was launched asynchronously.

You can also see this under the web system settings and the definition can be changed from there.