silvanmelchior/RPi_Cam_Web_Interface #442: new video files are: "busy"...?

T0m3kN51766, 2018-08-31:
I mounted a 128 gb thumb drive and bound it to my primary 32 micro sd card containing rasbian OS. After several recordings, all of my motion alerts continue to go off but videos are not recording.  I cannot view the video files. it's just a snapshot, roughly 32 mb per video file, and it says "busy"... 

What's going on here? I'll try to get a screen shot. I can't get one right now since I just deleted all of my video files and everything is working properly now. I think it'll all go down the drain after I use up all of the space on my 32 gb micro SD... I guess this software doesn't want to transfer over to the usb bound memory? has anybody had this problem before?

roberttidey, 2018-08-31:
The software will work with a media folder positioned elsewhere; USB, stick, USB drive, network location assuming it is mounted (fstab).

The best way to do this is to link the new folder on the new device to the normal SD media location (e.g. /var/www/html/media. This can be done with ln command.

The new location should also be owned by user www-data (chown) and have read write permissions.

For locations like network storage that may have problems supporting the video stream (network) there is a boxing path mode that allows the real time recordings to be made to the SD card and then moved during the MpBox conversion to .mp4 

The web will show busy if it can't find a mp4 file corresponding to the thumbnail. This can be because the video recording (h264) didn't work or because the MpBox conversion to .mp4 didn't work. Either way the log in scheduler screen will show what is going on.

icarito, 2018-09-19:
All the logs say is this:
{2018/09/19 05:22:21} Add /var/www/cam/media/vi_0022_20180919_052214.mp4 to Box Queue at pos 22

I'm experiencing the same issue. I installed ffmpeg but I don't think that's what's missing. Is there a dependency I'm missing (e.g. gstreamer?).

Thanks!

icarito, 2018-09-20:
The issue was missing MP4Box program. It comes in the `gpac` package! Install gpac and videos get converted when recorded!

roberttidey, 2018-09-20:
Good to hear it is working. The install does try to install gpac. That must have failed for some reason. Sometimes the repositories are unavailable for a short period.

BETLOG, 2023-01-16:
I have been seeing quite a few 'Busy' recently, for no apparent reason.
I ended up just telling the pi to queue them again.
`ssh pizerocam3 "bash /var/www/html/macros/startstopX.sh start"`

roberttidey, 2023-01-16:
Busy files do indicate that the MP4Box command used to convert to Mp4 has failed for some reason. A common cause of this it that the media folder is on a storage location that is not available temporarily for some reason.

The reason why the files that you have converted are not showing as completed is probably down to the ownership and permissions of the newly converted mp4 files. The web server can only see files which belong to the www-data user. This is for security reasons to prevent hacking.

You can fix the ownership of files you have converted by using the chown command.

The startstop script is designed to be run by the web server at start of the camera software and when it does the conversion the files will have the correct ownership. To activate it just rename the example startstopX.sh macro and check it is owned by www-data and has execution privileges.