silvanmelchior/RPi_Cam_Web_Interface #443: Broken image icon instead of live-view

mathue-git, 2018-10-02:
Hello,
Instead of a live-view of the camera, I see a constant updating "broken-image" symbol on the web interface. When I inspect the icon it says image file can not be found. 
The /dev/...imagejpg  (I can not recall the complete path - found it somewhere in the FAQ) does never contain any images but a constantly updating one .part file.

My setup:
- Raspberry Zero W 
- Pi NoIR Camera V2
- Latest Raspian Desktop img 

I tried reinstallating with apache and lighttpd - same result
I tried firefox and chrome as clients - same result 
I updated and upgraded via apt-get 

Everything else seems to work as expected - e.g. after pressing the [stop camera] button the broken image icon dissapears. Everything autostarts after reboots. 

any ideas?

roberttidey, 2018-10-02:
If you are not getting an image or a preview then it sounds like the raspimjpeg process is not managing to open a connection to the camera correctly. The log file (in the scheduler settings page) may give some indication of what is going on).

With the the raspimjpeg process not running (use ./stop.sh command in the RPi_Cam_web_Interface folder where you installed from) then try using raspistill to see if you can grab a still image OK.

If you can't then you need to check your camera connections to the Pi Zero. Make sure it is pushed in fully at both ends and also check the little ribbon cable on the camera module itself (Press it down).

Also check the power supply arrangements. The camera takes more power and so the 5V supply must be good. Some USB power cables are not very good and you can get significant voltage drop along the cable. It is best to use a short thick cable.

mathue-git, 2018-10-02:
What I did:
- reinstalled with raspbian light - so on CLI now 
- replaced the power supply (I use a dedicated raspberry power supply)
- double checked the ribbon connection - contacts facing on the board on each side ... fully pushed back (ribbon and connectors) 
still no picture. 

I used stop.sh and tried to take a picture with raspistill :

> mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board

So I guess this doesn't look good for my camera module :(

Where exactly can I check further log files?
Does the Pi NoIR Camera V2 need any further settings/driver ?

roberttidey, 2018-10-03:
You can access the log files from the scheduler settings page. This allow you to view, download and purge the log file.

If raspistill is giving that error message then there is something definitely wrong with pi getting at the camera module and the RPi_Cam software will not work. There is no electrical / software difference between NoIR and ordinary camera. It is just whether the IR filter is in place over the lens.

The reference to the Sunny connector is to the tiny ribbon connecting the camera to the module. Sometimes that can get loose and pushing it down should correct that.

If you have access to another Pi then I would check the module on that. Note also the camera csi connector on the Zero is fairly fragile and it is possible the plastic retainer has got broken and is not applying full pressure on the ribbon. Some people have corrected this by using tape.

mathue-git, 2018-10-09:
I checked the camera on a different pi - same problem. I ordered a new module and it worked instantaneous.
Thank you for the troubleshooting.

closed 2018-10-09 by mathue-git