silvanmelchior/RPi_Cam_Web_Interface #445: No timelapse conversion

tmo26, 2018-10-12:
Debian Stretch /  RPi Cam Control 6.4.30 /  ffmpeg 3.2.10-1~deb9u1+rpt2

**Issue:** Timelapse videos are not being created

**Possible rootcause:** Missing space in the ffmpeg command

- approx 2500 pictures in the timelapse series
- Click _Start Convert_
- Log shows: 
```
[2018/10/12 20:00:42] start lapse convert:(/usr/bin/ffmpeg -f image2 -framerate 15 -i /var/www/html/picam/media/t0008/i_%05d.jpg -c:v h264_omx -r 15 -b:v 2500k/var/www/html/picam/media/tl_0008_0001_20181011_223929.mp4 ; ) >/dev/null 2>&1 &
[2018/10/12 20:00:42] Convert finished
```
Finishing 2500 pictures at the very same second the conversion command started? Impossible. No video has been created.
- If I run the command from the log above manually on the console with an extra space between 2500k and /var/www/... it works, video is being created.
`/usr/bin/ffmpeg -f image2 -framerate 15 -i /var/www/html/picam/media/t0008/i_%05d.jpg -c:v h264_omx -r 15 -b:v 2500k /var/www/html/picam/media/tl_0008_0001_20181011_223929.mp4`
- My /var/www/html/picam/convertCmd.txt:
`/usr/bin/ffmpeg -f image2 -framerate 15 -i i_%05d.jpg -c:v h264_omx -r 15 -b:v 2500k`
Adding that extra space to convertCmd.txt does not result in the extra space being present in the log output.

Can you please fix the missing space in the ffmpeg command?

pishotta, 2018-10-12:
Hi,

I ran into something similar some time and several versions ago.  It had to do with the need of the software to preserve a trailing space at the end of the string in convertCmd.txt.  It was apparently being trimmed.  I don't recall whether this happened while storing or while retrieving the string, but it was one or the other. This made no difference in the default gstreamer setup where a directory/file name was being immediately appended to a string ending with an "=", but would cause processing to fail where a &lt;space&gt; was required before the directory/file name by software using a different command syntax.  Maybe this is the same thing.  If so, perhaps Robert can address it.

Because I was having other problems with gstreamer, I eventually moved the boxing processes to a script of my own, which avoided convertCmd.txt entirely.

-- Fred P.

roberttidey, 2018-10-12:
I don't know why the trailing space is getting stripped as there is no explicit trim command in there. It might have been caused by the switch from php5 to php7 some time ago.

I have added a space in explicitly between the convertCommand.txt template and the rest of the line.

ffmpeg now converts OK. I will do some more checking of the gstreamer again as this was the preferred method.

tmo26, 2018-10-14:
I can confirm, that after updating RPCWI, ffmpeg now converts ok.