silvanmelchior/RPi_Cam_Web_Interface #454: Full screen permanently

rumplepusscat, 2018-11-27:
Hi,

I'm currently assisting someone who's using your interface and they currently have it set up so that you click on the live camera feed, and it goes fullscreen. However, we need it to be permanently fullscreen (i.e. to fill an iframe when we embed it externally). Since we're on a tight deadline and I haven't used your interface personally before, I thought I'd come directly to you here!

Many thanks,

Rumplepusscat

roberttidey, 2018-11-27:
The toggle works by calling the javascript function toggle_fullscreen(e) where e is the preview element object.

The action of the function is to set the classname to either "fullscreen" or "" The  background.style.display is also set to block or none.

There is an optional config attribute (fullscreen) that determines what the start up condition is. See wiki under basic usage or config file
E.g
fullscreen true 

in /etc/raspimjpeg config file then it will start up in fullscreen mode.

To prevent the toggle working either remove the onclick call to toggle_fullscreen in index.php or redefine the javascript toggle_fullscreen function so that it does nothing.

joshcarz, 2018-12-22:
I tried setting the "fullscreen true" option in the config file and it didnt work. Is this confirmed to work?

roberttidey, 2018-12-23:
Yes. It does work.
You can enter in one of two places
/etc/raspimjpeg

or in /var/www/html/uconfig

For the second one you may need to change the folder if installed somewhere else.

The uconfig file contains config differences applied by the user.

roberttidey, 2018-12-24:
Theoretically an update should not have been required. However, it will have done 2 things. It will have ensured the latest version was installed and it will have restarted the camera system.

It is possible you had a version where fullscreen was not supported, and the restart will have ensured the config was loaded.