silvanmelchior/RPi_Cam_Web_Interface #455: Manual capture for set number of seconds isn't working

tarcus69, 2018-11-29:
When I echo "ca 1 20" to the FIFO1 it triggers a manual capture, but only of the buffer, no additional video is recorded, so I get about 3-6 seconds of video, not the expected 20 seconds. I've tried various values for the capture seconds up to 2000000 but it's always just the buffer that gets recorded. Am I doing something wrong or does this not work?

The system was newly built last night from the latest version in git.

```
pi@camoffrear:~ $ echo 'ca 1 20' > /var/www/FIFO1
pi@camoffrear:~ $ echo 'ca 1 20' > /var/www/FIFO1
pi@camoffrear:~ $ echo 'ca 1 200' > /var/www/FIFO1
pi@camoffrear:~ $ echo 'ca 1 2000' > /var/www/FIFO1
pi@camoffrear:~ $ echo 'ca 1 5000' > /var/www/FIFO1
pi@camoffrear:~ $ echo 'ca 1 50000' > /var/www/FIFO1
pi@camoffrear:~ $ echo 'ca 1 50000' > /var/www/FIFO1
```

roberttidey, 2018-11-29:
FIFO not FIFO1 is the pipe for commands.

FIFO1 is used by the scheduler to handle motion detection events

I just tried echo 'ca 1 20' >/var/www/FIFO and it is working fine.

Note that is with an install directly into /var/www root.

tarcus69, 2018-11-29:
Ah right, sorry I got the FIFO1 path from the scheduler page where the command reference is and it accepted recording commands, so thought that was the right one. I'll amend my notes. I've just tried it and you're right, it works, so that's sorted. Thanks again.