silvanmelchior/RPi_Cam_Web_Interface #456: Upload to google drive.

andreluisos, 2018-12-05:
Is there a way to upload files after box? If not natively, could you guys recommend a script, please?

roberttidey, 2018-12-05:
I think rclone would be the best way to proceed here.

You can set that up to sync a folder on raspberry to google drive folder.

You could either just sync the media folder itself or if you want to be more selective then use the box script to copy the captured files you want to the local rclone folder

https://rclone.org/

andreluisos, 2018-12-05:
Is there a way to make it (rclone) run after box? As far as I understood, I should make a shell script file called "end_box.sh" and give it permissions. Is it right?

roberttidey, 2018-12-06:
Yes endbox.sh can be used. You need to make sure it is owned by user www-data (chown) and given execution permissions.

If you only wanted to sync at intervals you could use the management_command facility in scheduler. Say set to run every 30 minutes.

closed 2018-12-06 by andreluisos