silvanmelchior/RPi_Cam_Web_Interface #463: Image Statistics option causes saving black images with only annotation showing

slabua, 2018-12-30:
I have realised that at some point between July and August, the images saved appear to be all black with only the annotation showing up (if set up). The thumbnails, though, appear to be fine. Videos saved also appear to be fine, both preview and actual video).
I have just updated to the latest raspimjpeg binary, either from upstread (5.8.86) and this repo (5.8.11), thinking that might have been my issue, but the issue seemed to persist.

After looking at the settings, I have discovered that turning on the Image Statistics option is the culprit for this issue.
Could anyone reproduce?

roberttidey, 2018-12-30:
Just tried it out and it seems OK on my set up.

It is possible that it is dependent on other camera settings. If you could post your uconfig file from the installed web folder that would allow a better check against what other settings you are using.

roberttidey, 2018-12-31:
I'll check out for any effect of those settings in a couple of days.

You can copy your uconfig file then try it out incrementally. E.g start just with annotation.

You need to stop / start the camera from the web page to get it to reload any manually changed uconfig file.

slabua, 2019-01-01:
@roberttidey thanks I'll check it that way too.
And a happy new year!

closed 2019-05-20 by slabua