silvanmelchior/RPi_Cam_Web_Interface #469: Dynamically enable/disable internal motion mask file?

hazardlabs, 2019-01-14:
Hi there,

I've currently got a working setup, using a Pi Zero and the internal motion detection using a mask.

Is there a way to use the mask to trigger, but as soon as the trigger condition is met, to use the full screen (i.e switch off the mask) until no more movement is detected. Then reset?

I've got my camera pointed at my drive way, with a mask so the internal motion detection only triggers when things happen on the driveway, however it would be beneficial to be able to see what happens when people/objects leave the area and what they get up to. Is this possible to achieve?

roberttidey, 2019-01-14:
Let me check I understand this correctly.

The mask does not affect the video recorded only the region used to calculate whether movement is happening.

So is it that you want to use the mask when calculating a movement start trigger but then use the full screen when calculating the conditions for a stop? You can extend the recording by setting the still frames to stop to a larger number but this is still using the masked region.

Currently when a mask file is defined then it is used all the time. It would be possible to have a separate command to enable the use of the mask on and off. This could then be used to disable it when a recording starts and re-enable it when it stopped. This does require a software change but should be fairly straightforward if that is what you want.

roberttidey, 2019-01-14:
Latest version has mz command to disable the mask dynamically. So ca 1;mz 1  will start recording and disable the mask. Similarly ca 0;mz 0 will stop recording and enable mask

roberttidey, 2019-01-15:
Yes. In the scheduler settings the motion start and motion stop fields determine what action to take when motion triggers occur. So ca 1 and ca 0 just start and stop video recording. You can put in a whole sequence of commands in these fields separated by ';'

hazardlabs, 2019-01-15:
Thank you for the clarification.

I've updated my settings and report back on how it does by the end of the week, give it a good soak test.

Are there any log files you want me to keep track of during this?

hazardlabs, 2019-01-16:
It appears that the mask disable, mz 1 works well, however mz 0 doesn't appear to re-enable it.

The text "Disabling Mask" is present, but no "Enabling Mask

Log below:

{2019/01/16 12:34:33} Capturing stopped
{2019/01/16 12:34:33} Add /var/www/media/vi_0009_20190116_123425.mp4 to Box Queue at pos 16
{2019/01/16 12:34:33} Start boxing /var/www/media/vi_0009_20190116_123425.h264 to /var/www/media/vi_0009_20190116_123425.mp4 Queue pos 16
[2019/01/16 12:34:35] Send mz 0
{2019/01/16 12:34:35} Finished boxing /var/www/media/vi_0009_20190116_123425.mp4 from Box Queue at pos 16
{2019/01/16 12:34:35} Removed item from Box Queue
{2019/01/16 12:34:35} Executing macro /var/www/macros/motion_event.sh "1" &
{2019/01/16 12:34:35} send smd 1
[2019/01/16 12:34:37] Start capture requested from Pipe
[2019/01/16 12:34:37] Send ca 1
{2019/01/16 12:34:37} Capturing started
[2019/01/16 12:34:39] Send mz 1
{2019/01/16 12:34:39} mask disable 1
{2019/01/16 12:34:47} Executing macro /var/www/macros/motion_event.sh "0" &
{2019/01/16 12:34:47} send smd 0
[2019/01/16 12:34:47] Stop capture requested
[2019/01/16 12:34:47] Send ca 0
{2019/01/16 12:34:47} Capturing stopped
{2019/01/16 12:34:47} Add /var/www/media/vi_0010_20190116_123437.mp4 to Box Queue at pos 17
{2019/01/16 12:34:47} Start boxing /var/www/media/vi_0010_20190116_123437.h264 to /var/www/media/vi_0010_20190116_123437.mp4 Queue pos 17
[2019/01/16 12:34:49] Send mz 0
{2019/01/16 12:34:52} Finished boxing /var/www/media/vi_0010_20190116_123437.mp4 from Box Queue at pos 17
{2019/01/16 12:34:52} Removed item from Box Queue

hazardlabs, 2019-01-16:
I had to reinstall my setup, not upgrade as I did before. It seems to be working again now, with the log posting "mask disable 0" as expected. 

I'll advise in a few days time if it falls over again.

Thank you for your help, it is very much appreciated!

hazardlabs, 2019-01-17:
And its back to the previous behavior, Here is my full log, minus username

You can see at the begining that it correctly re-enables the mask, but further on it does not.

[2019/01/16 20:33:02] Start capture requested from Pipe
[2019/01/16 20:33:02] Send ca 1
{2019/01/16 20:33:03} Capturing started
[2019/01/16 20:33:04] Send mz 1
{2019/01/16 20:33:05} mask disable 1
[2019/01/16 20:33:06] Send sy email.sh
{2019/01/16 20:33:07} Executing macro /var/www/macros/email.sh &
{2019/01/16 20:33:10} send smd 0
[2019/01/16 20:33:10] Stop capture requested
[2019/01/16 20:33:10] Send ca 0
{2019/01/16 20:33:11} Capturing stopped
{2019/01/16 20:33:11} Add /var/www/media/vi_0000_20190116_203303.mp4 to Box Queue at pos 0
{2019/01/16 20:33:12} Start boxing /var/www/media/vi_0000_20190116_203303.h264 to /var/www/media/vi_0000_20190116_203303.mp4 Queue pos 0
[2019/01/16 20:33:12] Send mz 0
{2019/01/16 20:33:12} mask disable 0
{2019/01/16 20:33:13} Finished boxing /var/www/media/vi_0000_20190116_203303.mp4 from Box Queue at pos 0
{2019/01/16 20:33:13} Removed item from Box Queue
[2019/01/16 20:34:03] Saved schedule settings
[2019/01/16 20:34:03] Send Schedule reset
[2019/01/16 20:34:03] Reload parameters command requested
[2019/01/16 20:34:03] Scheduler loop is started
[2019/01/16 20:34:04] New period detected 1
[2019/01/16 20:34:04] Send em night
{2019/01/16 20:34:04} Change: exposure_mode = night
[2019/01/16 20:34:06] Scheduled management tasks. Next at 1547674444
{2019/01/16 20:34:48} send smd 1
[2019/01/16 20:34:48] Start capture requested from Pipe
{2019/01/16 20:36:42} send smd 0
[2019/01/16 20:36:42] Stop capture request ignored, already stopped
{2019/01/16 20:51:06} send smd 1
[2019/01/16 20:51:07] Start capture requested from Pipe
{2019/01/16 20:53:07} send smd 0
[2019/01/16 20:53:07] Stop capture request ignored, already stopped
[2019/01/16 21:34:05] Scheduled management tasks. Next at 1547678045
[2019/01/16 22:34:06] Scheduled management tasks. Next at 1547681646
[2019/01/16 23:34:07] Scheduled management tasks. Next at 1547685247
[2019/01/17 00:00:08] New period detected 1
[2019/01/17 00:00:08] Send em night
{2019/01/17 00:00:08} Change: exposure_mode = night
[2019/01/17 00:34:08] Scheduled management tasks. Next at 1547688848
[2019/01/17 01:34:09] Scheduled management tasks. Next at 1547692449
[2019/01/17 02:34:10] Scheduled management tasks. Next at 1547696050
[2019/01/17 03:34:11] Scheduled management tasks. Next at 1547699651
[2019/01/17 04:34:12] Scheduled management tasks. Next at 1547703252
[2019/01/17 04:58:03] New period detected 2
[2019/01/17 04:58:03] Send md 1
{2019/01/17 04:58:03} Internal motion detection started
{2019/01/17 04:58:03} h264 search /var/www/media/vi_0001_20190117_045803.mp4
{2019/01/17 04:58:03} recommended video buffer size 65536
{2019/01/17 04:58:03} h264 size set to 131072
{2019/01/17 04:58:03} recommended video buffers 1
{2019/01/17 04:58:03} h264 buffers set to recommended 1
{2019/01/17 04:58:03} Set up internal detect width=82 height=61
{2019/01/17 04:58:03} Motion mask /var/www/mask.pgm loaded
[2019/01/17 04:58:05] Send em night
{2019/01/17 04:58:05} Change: exposure_mode = night
{2019/01/17 04:58:06} send smd 1
[2019/01/17 04:58:07] Start capture requested from Pipe
{2019/01/17 05:00:05} send smd 0
[2019/01/17 05:00:05] Stop capture request ignored, already stopped
[2019/01/17 05:34:13] Scheduled management tasks. Next at 1547706853
{2019/01/17 06:02:15} send smd 1
[2019/01/17 06:02:15] Start capture requested from Pipe
{2019/01/17 06:04:12} send smd 0
[2019/01/17 06:04:12] Stop capture request ignored, already stopped
[2019/01/17 06:34:14] Scheduled management tasks. Next at 1547710454
[... 72 lines cut ...]
[2019/01/17 09:12:31] Stop capture requested
[2019/01/17 09:12:31] Send ca
{2019/01/17 09:12:32} Capturing stopped
{2019/01/17 09:12:32} Add /var/www/media/vi_0003_20190117_091222.mp4 to Box Queue at pos 3
{2019/01/17 09:12:32} Start boxing /var/www/media/vi_0003_20190117_091222.h264 to /var/www/media/vi_0003_20190117_091222.mp4 Queue pos 3
{2019/01/17 09:12:33} Finished boxing /var/www/media/vi_0003_20190117_091222.mp4 from Box Queue at pos 3
{2019/01/17 09:12:33} Removed item from Box Queue
[2019/01/17 09:12:33] Send mz 0
{2019/01/17 09:22:20} send smd 1
[2019/01/17 09:22:20] Start capture requested from Pipe
[2019/01/17 09:22:20] Send ca 1
{2019/01/17 09:22:20} Capturing started
[2019/01/17 09:22:22] Send mz 1
{2019/01/17 09:22:22} mask disable 1
[2019/01/17 09:22:24] Send sy email.sh
{2019/01/17 09:22:24} Executing macro /var/www/macros/email.sh &
{2019/01/17 09:22:27} send smd 0
[2019/01/17 09:22:27] Stop capture requested
[2019/01/17 09:22:27] Send ca
{2019/01/17 09:22:27} Capturing stopped
{2019/01/17 09:22:27} Add /var/www/media/vi_0004_20190117_092220.mp4 to Box Queue at pos 4
{2019/01/17 09:22:28} Start boxing /var/www/media/vi_0004_20190117_092220.h264 to /var/www/media/vi_0004_20190117_092220.mp4 Queue pos 4
[2019/01/17 09:22:29] Send mz 0
{2019/01/17 09:22:29} Finished boxing /var/www/media/vi_0004_20190117_092220.mp4 from Box Queue at pos 4
{2019/01/17 09:22:29} Removed item from Box Queue

roberttidey, 2019-01-17:
Have you got ca 0 for the stop? It looks like you just have ca. That may be causing the pipe commands to get out of step.

hazardlabs, 2019-01-17:
Robert,

Good catch, although still the same result.

I've saved all the settings and done a systems restart, still with the same effect.

[log.txt](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/2768700/log.txt)

roberttidey, 2019-01-17:
I'm not seeing the problem in that log file.

There are 8 recordings; all have mz 1 after starting and mz 0 after stopping.

hazardlabs, 2019-01-17:
The latest update:

In the scheduler, if I use the command 
`ca 0;mz 0`
 it starts to fall over. If I use 
`ca 0; mz 0;`
with the extra semi colon at the end instead its working more reliably.

roberttidey, 2019-01-17:
There is a gap of 2 seconds inserted into each command of a semicolon list. By putting a semicolon on the end the scheduler will wait for 2 seconds before processing any further events.

I noticed in some of your log data that you are getting new motion starts very soon after one stops. It is possible that one following very quickly after the mz 0 is sent is overlapping in the FIFO pipe and preventing that being processed correctly. If so then the extra delay would prevent that happening,

I'll check that out further. You could also try increasing the Cmd_Poll setting in the scheduler to say 0.1 This will help prevent possible command overlap.

hazardlabs, 2019-01-18:
I had also noticed the rapid re-triggering as well.

I've upped the stable frames from 150 to 250, with the same result.

I'll update the cmd_poll and report back

hazardlabs, 2019-01-18:
The change to cmd_poll has done the trick, in making the 'mask disabled 0' appear in the log file every time so far.

The mask appears to be enabled as my notifications and recordings from the system have dramatically reduced as required.

hazardlabs, 2019-01-19:
System still operating as required, I get some videos that trigger too close together, but I believe that is a separate video at this point. 

Issue resolved by changing cmd_poll to 0.01

Closing Issue.

closed 2019-01-19 by hazardlabs