silvanmelchior/RPi_Cam_Web_Interface #470: Motion Settings: Defaults appear swapped

hazardlabs, 2019-01-18:
Looking at the Motion Settings tab on the main page, it appears the values for noise level and Threashold are in the wrong places. Please see the attached image

[image]

Unless the title descriptions next to them are wrong?

roberttidey, 2019-01-18:
The parameters are the right way round but the commented range for Noise level is incorrect. 0-255 refers to a simple algorithm. Noise values above 1000 switch to a newer (better) algorithm. See in wiki under internal motion detection for a more detailed explanation.

I have amended the comment to help avoid confusion.

hazardlabs, 2019-01-18:
Perfect, Thanks. Closing Issue

closed 2019-01-18 by hazardlabs