silvanmelchior/RPi_Cam_Web_Interface #472: sorting issue on "download videos and images" page

wingers999, 2019-01-20:
Hi

Seem to only recently have issue with how items sorted on the download page - it used to be in date/time order, but now it seems completely muddled, despite sort ascending being selected

Screenshot shows what I mean better than I can explain via writing

[image]

roberttidey, 2019-01-20:
Your screenshot has 1 image plus 3 time lapse sets plus a bunch of videos. The page orders things as sorted images, followed by sorted time lapse, followed by sorted videos.

This has not changed. Note you can use the types filter to just show videos.

wingers999, 2019-01-20:
Thanks Robert - never noticed it did that before!

closed 2019-01-20 by wingers999