silvanmelchior/RPi_Cam_Web_Interface #473: Update appears to overwrite /usr/local/.htpasswd

ras07, 2019-01-23:
v6.4.xx on Apache, Stretch Lite, Pi Zero W. I've added several users per the instructions at https://elinux.org/RPi-Cam-Web-Interface#Add_extra_users_to_security and https://elinux.org/RPi-Cam-Web-Interface#User_Access_levels:_Simple_user_access_control_of_web_display_functions . After an update, /usr/local/.htpasswd contains only one line, reflecting the user configured in the Configuration Options during the update.

roberttidey, 2019-01-23:
Yes. The htpasswd operation in the install script used the -c flag which creates a fresh copy of the .htpasswd file. One has to use the -c flag at least once to create the file.

I have added now some extra logic in the install which should respect a .htpasswd file and therefore preserve the passowrds of any additional users. It will update the passwd for the one user configured in the set up.

ras07, 2019-01-24:
Great, I just upgraded to 6.4.41 and it preserved the existing users - Thanks! I'll close the issue.

Thanks again!

closed 2019-01-24 by ras07