silvanmelchior/RPi_Cam_Web_Interface #474: end_vid macro is passed .mp4 filename when boxing turned off

nmilnercollabora, 2019-01-24:
I've turned off mp4 boxing as intend to process the h264 files at a later time using a PC. I do however use the end_vid macro to move the h264 files to an external drive. The end_vid macro is passed the video filename however, it assumes the video is in the mp4 container. I currently work around this by changing the file extension passed to the macro from .mp4 to .h264 but it would helpful if the filename passed to the macro when boxing is off is .h264.

roberttidey, 2019-01-24:
end_vid macro now called with .h264 name for both boxing on or off

jarrah31, 2019-04-06:
Sorry guys, I was relying on end_vid passing the video filename as the parameter in my script, but this update has now broken it for me. I've only just noticed as it's been a while since I've had to rebuild the camera software.

It was originally using this:
`Executing macro /var/www/html/macros/end_vid.sh "/var/www/html/media/2019/04/06/vi_14704_20190406_080005.mp4" &`

But is now this:
`Executing macro /var/www/html/macros/end_vid.sh "/var/www/html/h264/vi_8907_20190406_140756.h264" &`

I need the full video file path so that I can copy the converted mp4 file to the correct dated folder in /var/www/html/media (which is a remote mount point). 

I too was originally just obtaining the file name and converting the extension to .h264 because my script basically does the MP4 Boxing on the Pi itself. I don't use the built-in boxing because I have to do lots of checks to ensure the /var/www/html/media mount point is ok (I have historically found mount points can be flaky over wifi) and check that the copy worked before then deleting the local .h264 file.

I'd like to suggest that we revert this change please and that if we need .h264 passed as the filename to end_vid.sh, that we just change the `video_path` parameter in /etc/raspimjpeg instead? e.g.

`video_path /var/www/html/media/vi_%v_%Y%M%D_%h%m%s.mp4`

to
`video_path /var/www/html/media/vi_%v_%Y%M%D_%h%m%s.h264`

Thanks.

roberttidey, 2019-04-06:
The reason the path is changing is because you also have set up a boxing_path. This was intended to be the place where h264 files were stored prior to them being processed by the built in boxing process. This allowed the recording to be local (e.g. on SD card) and the boxing to process to the final media folder which could be a remote mount. The internal boxing process is much more tolerant of slow downs in networking than the real time video recording.

I think you could set the boxing_path to be almost the same as video_path but with media changed to h264. That would give you the same structure but the script would need to change the h264 back to media in calculating the final destination. However it might have some side effects in leaving empty folders under the boxing_path.  

I'll see if I can find a way of satisfying both requirements here. Just changing the video_path wouldn't work as the filename has to go through several transformations during the whole processing pipe.

jarrah31, 2019-04-06:
Thanks for looking into this Robert.

I currently have the following set
```
boxing_path /var/www/html/h264
video_path /var/www/html/media/vi_%v_%Y%M%D_%h%m%s.mp4
```
Are you saying that boxing_path can include the filename as well to look like this? 
```
boxing_path /var/www/html/h264/vi_%v_%Y%M%D_%h%m%s.mp4
```
If so then I don't mind working the script to delete date folders after converting h264 to mp4, as it would at least give me the date folder path to copy mp4 files into media.

roberttidey, 2019-04-06:
Yes. I think that would work.

I'm still seeing if there is a way to do this more cleanly and allowing the boxing_path to be flat.

Maybe if the video_path was always used but extension changed to h264.

If boxing path was null then the path passed to the end_script would directly reflect the source file h264 under media which maybe satisfies the original issue.

If boxing_path was set up then the path passed to end script would reflect the logical path but the source path would still need to be calculated under boxing_path as I guess you were doing anyway.

jarrah31, 2019-04-06:
oh yes sorry, video_path on another Pi that I set up to use folder dates looks like this:
```
video_path /var/www/html/media/%Y/%M/%D/vi_%v_%Y%M%D_%h%m%s.mp4
```
I'm just going to try using this on boxing_path now.

roberttidey, 2019-04-06:
Checking the code again I don't think that is going to work.

roberttidey, 2019-04-06:
I think my idea of reverting and adjusting just the extension may be a better approach.

I should be able to do something tomorrow on that.

jarrah31, 2019-04-06:
yup, it just creates a folder as it's written :)

jarrah31, 2019-04-06:
sounds good, thanks Robert!

roberttidey, 2019-04-07:
Latest version now passes to end_vid the path to file under media folder but with .h264 extension

jarrah31, 2019-04-08:
super, thanks Robert! I'll try it out this evening.

jarrah31, 2019-04-08:
works great, thank you!