silvanmelchior/RPi_Cam_Web_Interface #477: FIFO commands don't work when called from macro scripts

ras07, 2019-02-07:
Commands issued to the FIFO queue don't seem to work as expected when called from a macro script. For example, with this: 
`#!/bin/bash`
`echo 'md 0' > /var/www/html/FIFO`
`sleep 60`
`echo 'md 1' > /var/www/html/FIFO`
in end_box.sh, I would expect motion detection to be turned off for 60 seconds after a video is recorded, and then get turned back on. This script works fine when executed manually, but when triggered by the completion of a boxing event, it doesn't work right. Often the first command will execute but not the second, or what appears to be a partial command will execute (for example if I use 
`echo 'an ABCDEFG' > /var/www/html/FIFO`
for testing purposes, the annotation will sometimes be changed to 'AB').
Script is owned by www-data and permissions are 744. The script works fine if there are no FIFO commands in it, so I don't believe it is a permissions or ownership problem.

roberttidey, 2019-02-07:
I don't think your issue is commands not working from macro scripts.

From the sound of it you are occasionally getting partial commands. This may be happening if you have a heavily loaded Pi particularly if it is single core.

A bit of background. Originally FIFO commands were processed as soon as they arrived and didn't have  any line terminator. This could cause problems if 2 commands were sent really quickly as the whole of the buffer was being read as 1 command. This was enhanced to allow LF terminators and the commands split up and processed correctly. The older method was left in place for backwards compatibility to avoid breaking programs that didn't send LF.

So the behaviour is currently commands get processed whenever LF is found OR if more than 100mSec elapses without any FIFO input.

This means that if your script gets interrupted by a process for more than 100 mSec then a partial command could get processed.

I am looking into introducing an option which would enforce using LF which would avoid that problem, but I would have it default to the current behaviour. I'll update when this is available.

Note that echo does include a LF terminator (echo -n does not include LF)

roberttidey, 2019-02-07:
The current version now has 2 new config parameters in /etc/raspimjpeg.

enforce_lf when set to 1 will only process fifo commands when a terminating lf is received.

fifo_interval allows the fifo polling interval to be increased from its current 100000 microseconds. It should not be necessary to do this if using enforce_lf

ras07, 2019-02-07:
`enforce_lf 1` seems to solve the problem - thanks! I'll close the issue.

closed 2019-02-07 by ras07