silvanmelchior/RPi_Cam_Web_Interface #478: update.sh fails

ttfna, 2019-02-07:
have a successful install of RPi_Cam but the update is failing with the following:

────────Update message───────────┐
                                                      │ Commits don't match. We update. │  

error: rev-list died of signal 11
remote: Enumerating objects: 181, done.
remote: Counting objects: 100% (181/181), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 326 (delta 163), reused 166 (delta 151), pack-reused 145
Receiving objects: 100% (326/326), 225.56 KiB | 0 bytes/s, done.
Resolving deltas: 100% (221/221), completed with 23 local objects.
error: rev-list died of signal 11
error: https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git did not send all necessary objects

***************
*** ABORTED ***
***************

An error occurred. Exiting...

Any help appreciated

roberttidey, 2019-02-07:
Sometimes that happens if something has changed in the current install folder and git can't resolve the differences.

Just wipe the install folder  (RPi_Cam_Web_Interface) and do a normal install i.e clone etc). This has the same effect as an update anyway.

ttfna, 2019-02-07:
thanks. that and running:

sudo apt-get install git

before the update fixed it

closed 2019-02-07 by ttfna