silvanmelchior/RPi_Cam_Web_Interface #480: Motion detection

mtasilva, 2019-02-12:
Hello, i use raspberry pi zero w with stretch system.
I install RPI_Cam_Web_Interface, with a infrared cam.
When i use start motion detect , sometimes, he detects and save video, sometimes, just no detect and can stop detection motion and de image gets blocked, its aleatory ...

roberttidey, 2019-02-13:
Are you using internal or external motion detection? I recommend using internal; it works better and is easier to tune up to suit the type of scene the camera is seeing.

If you include %f and %c in the annotation string then that makes it easier to adjust the motion parameters. %c show the change value the detection is seeing per frame. This must on average get above the threshold for a frame to have 'motion in it and advance. %f gives the motion detected frame count. This must get above the start frame count to trigger detection. Similarly lack of motion once triggered will advance the frame count when no motion is detected and once the %f gets above the stop frame count it will trigger the end of motion.

mtasilva, 2019-02-14:
Hello, I started using it recently. The first time I installed it and it quickly seemed to work fine. 
After I did an update later.So..
Where i define internal or external motion detection? I can not find !! And where i define %f and %c?
My interface:

[image]

[image]

My cam is : CEWAAL Fisheyes Wide Angle Night Vision Camera Cam Vedio Infrared Infrared Fill Light
I do not use any Pir motion module with raspberry only camera.
My scenario is just to detect someone and send an email.
Can you help put de correct settings?
Thank you for your time.

roberttidey, 2019-02-14:
Down the bottom f your camera settings is motion detect mode. You have it set to external.

I recommend internal. When you select that the motion settings button disappears but is replaced by a motion settings under camera settings. Start with default values.

The %f and %c are just put into the annotation string near the top under camera settings. They get replaced by the two variables I described so you see those in real time on the camera video.

mtasilva, 2019-02-15:
Ohh, yes i can see ..
[image]

Last night, I experimented with the motion detect as internal and it worked fine.
During the day it still blocks ...
How do I adjust the parameters now?

Thank you for your time.