silvanmelchior/RPi_Cam_Web_Interface #481: File sizes roll over above 4 GB

ras07, 2019-02-18:
This is a minor thing, but in the Files listing on the preview.php page, files that are over 4 GB display as ([true file size] mod 2^32).

(Granted, letting files get that big is probably unwise to begin with)

roberttidey, 2019-02-19:
Yes. This a common issue onwith php based systems as it uses 32 bit integers and so rolls over >4GB.

There are nasty work arounds involving shelling out a ls command and then parsing the result to get the file size but I'm reluctant to do that as it could slow things down quite a lot.

As you say files > 4GB are not a good idea anyway. If one wants to do long continous recordings then it is best to use the video split facility to break it up into more manageable chunks. One can also lower the video bit rate to keep the file sizes smaller without losing significant quality.

ras07, 2019-02-26:
Turns out there's a workaround involving the stat command that is already implemented in **config.php**:
` // control how filesize is extracted, 0 is fast and works for files up to 4GB, 1 is slower
define('FILESIZE_METHOD', '0');
`
I tried it and it did not seem to noticeably slow down the preview screen, even on a Pi Zero W with a lot of files to list. 

For now I'm just going to split the files, but I'll post this for future reference in case someone has a need for this. Closing the issue.

closed 2019-02-26 by ras07

roberttidey, 2019-02-27:
Thanks. I'd forgotten I did that a couple of years ago!!