silvanmelchior/RPi_Cam_Web_Interface #483: New Functionality

danriches, 2019-02-26:
Hi Silvan,

Great work by the way! I'm looking to do something quite specific and wondered if your application could have some features added or if it's something that you could provide in a template form that I can 'bend' to my needs. I've lots of issues in trying to get my project working and the hardware works brilliantly however the software v4l2 and mmal side of things are giving me lots of problems. I'm trying to stream the video (720x576) via an ADV7280A-M codec plugged into the CSI port of the RPi. I also, while rtsp streaming, need to be able to capture video or capture a single jpeg frame via a pipe call or via a GPIO input. Is that something you could help me with or give me any pointers with please?

Kind Regards,

Dan

roberttidey, 2019-02-26:
Silvan doesn't work on this any more. It is mainly me (roberttidey).

I am not quite sure what functionality you are asking for. This software uses raspimjpeg which uses mmal (not v4l2) to provide a live mjpeg stream and simultaneously allow motion detection, image, and video capture. All this is controlled from a pipe command interface which gives full control over the functions and settings.

Normally this is used with the browser interface. The pipe commands are  easily available outside of the browser interface by just writing commands into it. If you want the mjpeg live stream outside of the browser interface that is also possible by using the url which provides the stream. Users for example have viewed in vlc by this method.

All this depends of course on the success of plugging the ADV7280A-M into CSI and controlling it OK from MMAL.

danriches, 2019-02-26:
Hi Rob,

I guess my question was a bit long and vague, sorry about that. I guess my first step should be to check that RaspiMJPEG works with the ADV7280A-M and then go from there.

I have some questions though:
What sort of frame rate can we expect from the streamed images?
Can it handle audio from ALSA? I have a SGTL5000 codec with a mic on it that appears as an ALSA device and works very well, I'd like to get the audio streaming along with the video too. Is that something that RaspiMJPEG can do or can be modified to do?

Many Thanks, Dan

roberttidey, 2019-02-26:
The frame rate of the streamed images can be anything up to the video frame rate the camera is set to. So if camera is set to 25 fps then the normal preview can be up to that.

There is a preview divider setting which allows the preview to be lower. So if set to 2 then preview rate is 12.5 and so on. The purpose is to lower the network bandwidth required for the stream independent of the frame rate that captures will be made at. Similarly the raw pixel size and the compression quality can be set differently.

There is not any native support for audio capture in raspimjpeg. There have been one or two requests for this, but the main usage has been for video, time lapse and still image recording. There is also the issue of whether to support it for preview or for captures. Of course, one would like both but they are two separate problems. Preview would probably be easier. Embedding in captures is complicated by getting reasonable synchronism, particularly as captures can be buffered so that a few seconds of pre-trigger video also gets included.

Note that raspimjpeg source is also open on github.