silvanmelchior/RPi_Cam_Web_Interface #486: RPi Cam Web Interface recording not working

MarkOude, 2019-03-02:
Hello everyone!

I have installed this software on my Raspberrypi with the latest distro from raspberrypi.org.
It works correctly, but i can't record any videos or Motion Detection. I use my Raspi with SSH so i can configure it without getting it outside of my Birdhouse... If I click on the record button, it isn't recording in any way. I tried it with echo 'ca 1' > FIFO, but it also doesn't change anything.
If I press record I see in my Putty terminal: "mmal: mmal_port_disable: port vc.ril.video_encode:out:0(H264)(0x1b77730) is no enabled", and that is every time i am pressing record.
Taking Pictures is working though.

I Installed the software and the whole system 3 times, and always the same error.

PS: If i try to enable Motion detection then an alert comes:"Error in RaspiMJPEG.Restart RaspiMJPEG...."
and on my Terminal is: "mmal:mmal_vc_component_enable: failed to enable component: ENOSPC"

I tried 

Sorry for this much information...
Thanks in advance!

roberttidey, 2019-03-02:
The ENOSPC error indicates that there is something wrong at the system level accessing the camera properly. For some reason still images are working but video not. The live preview is actually a stream of still images.

Check following things.

1) Ensure camera cable is well inserted at both ends. Also press down on the tiny cable on the camera module itself.

2) Is your power supply OK? This includes a decent USB cable (some can be high resistance. A marginal supply at the Pi can cause camera problems.

3) Are you sure you have nothing else installed? Installing motion separately for example can screw things up. You should just let the installer put in what is required.

4) Check using raspivid that you can get video recordings from the camera that way. Do not start up the Cam web software for this test. Only 1 client can attach itself to the camera. If necessary run the ./stop.sh script in the install directory.

MarkOude, 2019-03-03:
So, I changed the Power supply to an 2.1A one, didn't change anything.
I had another Picam with cable laying around, so i connected it. But it wouldn't change anything. I set the GPU Memory to 256, and the bitrate to something less big, but both didn't work.

I installed the Software via install.sh, and did nothing else, i made a dist-upgrade and an update, but i don't htink that is changing somthing. Some time later i did an update.sh because it didn't work...

I tried raspivid with MP4Box, so i could play it on my laptop, and it did work correctly, the image was clear and no error occured.

Question: I am using an IR switching camera with one IR attatchment, could  that be a problem?

What i forgot in my original Question is, that just before the error "mmal: mmal_port_disable: port vc.ril.video_encode:out:0(H264)(0x1b77730) is no enabled", there is another error: "mmal: mmal_vc_port_enable: falied to enable port vc.ril.video_encode:in:0(OPQV):EINVAL"

Thanks for your help!

roberttidey, 2019-03-03:
The EINVAL error is just another indicator that the mmal is having a problem with setting up the video encoding.

Although if raspivid is working then there is something very specific as raspimjpeg which runs the camera uses very similar access routines to the mmal layer.

I don't know what IR switching camera with IR attachment means. Do you mean the NoIR camera (that is just one with the IR filter removed? They are exactly the same. If not please give more details.

It is possible a very recent raspbian release broke something. I do test with major releases but maybe a recent upgrade could have affected something. I'll check that, but can't until tomorrow. Please post details of your OS.

Run cat /etc/os-release  to get OS

and run uname -a  to get kernel

MarkOude, 2019-03-03:
The camera i meant is not the NoIR, it's one like this one: https://amzn.to/2Ui67AN
So, the output from cat /etc/os-release is:
PRETTY_NAME="Raspiban GNU/Linux 9 (stretch)
Name="Raspiban GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"

and the uname-a Output:
Linux vogelhaus 4.14.79+ #1159 Sun Nov 4 17:28:08 GMT 2018 armv61 GNU/Linux

stewartoallen, 2019-03-03:
I had all sorts of problems with the pi camera until I added these lines to ```/boot/config.txt```
```
over_voltage=6
force_turbo=1
```

MarkOude, 2019-03-03:
I tried this, but it didn't work. But i've got another Error message though: "Error: CCould not open/create video-file". I also tried it with over_voltage=2 and 4...

But Thanks for you help!

MarkOude, 2019-03-03:
I just did an update.sh with php 5 and it works!
BUT: The MP4Boxing isn't working at all,but i found out that the h246 is 0 bytes big, and that is my next problem. But the record button is working though...
And the Motion detection is also NOT working...

MarkOude, 2019-03-05:
Hello!

I Just reduced the Video size from 1920:1440 to 1280:960, and suddenly everything works.

roberttidey, 2019-03-05:
I think that video size exceeds the capability of the GPU particularly for raspimjpeg which has multiple operations going on in the GPU. 1920 x 1080 which is standard HD is fine. However 16:9 formats restrict the FOV available from the camera. 1280 x 960 is quite a good choice for full FOV and reasonable resolution.