silvanmelchior/RPi_Cam_Web_Interface #487: Implementing on FTP?

GaryLittlemore, 2019-03-10:
If I was to configure FTP access on the below link would that have any impact on the Cam Web Interface?

I think adding FTP access would be a great extra to be able to access the pages and media quicker via FTP.

https://pchelp.ricmedia.com/setup-lamp-server-raspberry-pi-3-complete-diy-guide/3/

roberttidey, 2019-03-11:
There is no issue adding an ftp server on to the same machine as running this interface.
If the ftp server is then configured to publish the media directory then a standard ftp client like filezilla can then download files directly.

Note you can also use something like winscp to transfer files back and forwards between a windows machine and the raspberry pi. This can be more convenient than ftp. It is quite fast and more secure. I often use this when transferring large numbers of files from say a time lapse capture.

GaryLittlemore, 2019-03-11:
Thanks Robert, I've just added the FTP setup. A much easier way to access and download the media files.

I think it would be great to be part of the standard setup when installing the web interface.