silvanmelchior/RPi_Cam_Web_Interface #488: remove.sh throws error string

danindiana, 2019-03-11:
Attempting to run the remove.sh on a Raspberry pi. 

Running ./remove.sh results in this error: 
./remove.sh: 101: ./remove.sh: Syntax error: "(" unexpected (expecting ";;")

Not sure what bash syntax should be in this case?

roberttidey, 2019-03-11:
Thanks for reporting this.

I think the problem is the extra ; on that line. There was also some other issues on that line with the quotes identifying the packages to be uninstalled.

It should be OK now in the released version.

danindiana, 2019-03-11:
Thanks for the fast reply. I am attempting to remove so I can start over with a fresh install, but found the remove.sh script wasn't working. Otherwise it's really great software for the RPI system.

roberttidey, 2019-03-11:
I don't really use remove myself as if I want a totally clean install then I start from a fresh image.

However, if it's still not working then I'll run some further tests to see what else could be going wrong.

Xer0Point, 2019-04-01:
The case statement (clause 0) appears to be missing its second semicolon. Adding it should fix the issue and allow the script to run without issue.

roberttidey, 2019-04-02:
Thanks. I have added that now.