silvanmelchior/RPi_Cam_Web_Interface #489: Timelapse randomly stops and starts creating new sequence of images

furtula37, 2019-04-11:
The issue is present if used with scheduler, or without, makes no difference.
Timelapse randomly stops and stars with new set of images.
Would it be possible to increase the timeout value, or investigate the issue more in depth?

Here's part of my log file when it happens:
{2019/04/11 02:18:15} Capturing image
{2019/04/11 02:19:15} Capturing image
{2019/04/11 02:20:15} Capturing image
{2019/04/11 02:21:15} Capturing image
{2019/04/11 02:22:15} Capturing image
{2019/04/11 02:23:44} Image capture timed out
{2019/04/11 02:23:49} RaspiMJPEG Version 5.8.16
{2019/04/11 02:23:49} h264 search /var/www/html/media/vi_0002_20190411_022349.mp4
{2019/04/11 02:23:49} recommended video buffer size 65536
{2019/04/11 02:23:49} h264 size set to 131072
{2019/04/11 02:23:49} recommended video buffers 1
{2019/04/11 02:23:49} h264 buffers set to recommended 1
{2019/04/11 02:23:49} Opening FIFO 0 /var/www/html/FIFO 7
{2019/04/11 02:23:49} Opening FIFO 1 /var/www/html/FIFO11 8
{2019/04/11 02:23:49} MJPEG streaming, ready to receive commands
{2019/04/11 02:23:49} send smd 9
{2019/04/11 02:23:49} Starting command loop
[2019/04/11 02:23:49] Reload parameters command requested
[2019/04/11 02:23:49] Scheduler loop is started
[2019/04/11 02:23:50] New period detected 16
[2019/04/11 02:23:50] Send tl 1
{2019/04/11 02:23:50} Timelapse started
[2019/04/11 02:23:52] Scheduled management tasks. Next at 1554945830
{2019/04/11 02:24:50} Capturing image
{2019/04/11 02:25:50} Capturing image

roberttidey, 2019-04-11:
Image capture time out is not directly associated with time lapse.

The image stream being constantly produced is monitored by a watchdog and if this stops for a period then the watchdog restarts the software. One can change the watchdog time under camera settings but this shouldn't be required.

The watchdog triggering normally indicates something unstable in the underlying system.

You should check camera cabling, power supply and the USB cable connecting the power. 

Also do you have any other apps running on the system that might affect performance. If you run top from a command line then you can check how loaded the cpu is.

furtula37, 2019-04-12:
Yes, that does make sense, my system is a bit botched, with KODI running on top of retropie, and RPI cam in the background, but this is not used at all during the timelapse, and idle usage is 10% at that time.  
Logs say nothing at all, but i do have crossover period during midnight when this usually happens, so i might stay and check active processes, or for some indication of the issue.
Coworker also has RPI cam with same setup, and i just checked his timelapse, and works fine with one being around 12k images, so you are right.

It is possible to be on the HW side, but since issue almost always occurs also on midnight, i suspect KODI.
I will investigate further, thank you for the input.

furtula37, 2019-04-13:
Upgrade to raspbian stretch fixed the issue.

closed 2019-04-13 by furtula37