silvanmelchior/RPi_Cam_Web_Interface #491: Showing output on HDMI (only using RCWI as camera control)

nospam2k, 2019-04-16:
Maybe I'm missing something simple, but I just want to use RCWI to control the camera remotely (changing PiCamera settings) but I don't see a setting for turning on or off HDMI overlay preview. How can this be accomplished? I don't need motion, webcam, recording options ATM.

roberttidey, 2019-04-17:
I don't understand what you mean by HDMI overlay preview. This software does not do anything with HDMI.

It does make available a mjpeg stream which can be accessed by the browser interface as required. Is that what you are referring to?

If so then what are you trying to achieve with the camera? Do you want a different web interface that is control only? One could do that with a simple modification to the index.php to skip the preview frame. But then how is the camera being used and how will you know what the effect of changing camera settings is?

nospam2k, 2019-04-17:
This is in a conference room setting with an overflow area. I can see the HDMI monitor while adjusting the camera. I haven't looked through the code yet but you are saying in the index.PHP I can turn off the streaming and enable the raspicam preview (overlay is the video preview over the login screen, no GUI)?

roberttidey, 2019-04-17:
I am still don't understand your set up.

What app is displaying on the HDMI monitor?

With the raspberry pi camera everything is controlled via the MMAL layer and only one application has ownership of that at any one time.

So if you have some other raspberry camera application then trying to control its settings has to be done via whatever controls app has. You can't do it via this interface.

On the other hand if the main HDMI monitor is showing the browser view of this app then a second browser client to the same server can remotely control its settings as it is the same server based instance of the camera app involved. The secondary client in this case can show the preview as well or it could be removed if desired as previously mentioned.

nospam2k, 2019-04-21:
It's ok. I did some homework and found that some simple coding with picamera.py and flask did the job. I just needed a simple setup to control the camera through a web interface.

closed 2019-04-21 by nospam2k