silvanmelchior/RPi_Cam_Web_Interface #492: Stretch / raspimjpeg watchdog problem

florisjan78, 2019-04-19:
I upgraded to stretch Linux thepi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux. Using lighttpd 

Now the video stream is gone.

Did apt-get install php7.0, removed rpi cam control and installed latest again. The interface works, not responsive though. 

Output from raspimjpeg:

`RaspiMJPEG Version 5.8.17
h264 search /var/www/html/media/vi_0000_20190419_105455.mp4
recommended video buffer size 65536
h264 size set to 131072
recommended video buffers 1
h264 buffers set to recommended 1
Opening FIFO 0 /var/www/html/FIFO 6
Opening FIFO 1 /var/www/html/FIFO11 7
MJPEG streaming, ready to receive commands
send smd 9
Starting command loop
Watchdog detected problem. StoppingSIGINT/SIGTERM received, stopping
start monitoring for pid: 1246
{2019/04/19 10:54:55} start monitoring for pid: 1246
RaspiMJPEG Version 5.8.17` etc..

output from scheduler log:
`[2019/04/19 11:10:03] Logged in user::
[2019/04/19 11:10:03] UserLevel 6
{2019/04/19 11:10:06} RaspiMJPEG Version 5.8.17
{2019/04/19 11:10:06} h264 search /var/www/html/media/vi_0000_20190419_111006.mp4
{2019/04/19 11:10:06} recommended video buffer size 65536
{2019/04/19 11:10:06} h264 size set to 131072
{2019/04/19 11:10:06} recommended video buffers 1
{2019/04/19 11:10:06} h264 buffers set to recommended 1
{2019/04/19 11:10:07} Opening FIFO 0 /var/www/html/FIFO 7
{2019/04/19 11:10:07} Opening FIFO 1 /var/www/html/FIFO11 8
{2019/04/19 11:10:07} MJPEG streaming, ready to receive commands
{2019/04/19 11:10:07} send smd 9
{2019/04/19 11:10:07} Starting command loop
[2019/04/19 11:10:07] Reload parameters command requested
[2019/04/19 11:10:07] Scheduler loop is started
[2019/04/19 11:10:08] New period detected 0
[2019/04/19 11:10:08] Scheduled management tasks. Next at 1555668608`

Does this have anything to do with the
`h264 search /var/www/html/media/vi_0000_20190419_105455.mp4`
line? There are no files in media.

Thank you!

roberttidey, 2019-04-19:
The h264 search line is just the start up function that looks for any left over .h264 files that had not been converted. The filename is meaningless. It is just used to get the search path.

If there is no file in /dev/shm/mjpeg then you won't get a preview stream and the watchdog will kick in as the log shows.

Stop the camera software. If you have uConfig file in /var/www/html then move it somewhere else so it is not in that folder. That will ensure it starts up with defaults as defined in /etc/raspimjpeg.

Restart the camera software. 

If that doesn't help then double check the connections to the camera. As a test you can stop this software and try using raspivid to see if that makes a recording.

florisjan78, 2019-04-19:
Thanks for the tip. I tried taking a raspistill and got 'No data received from sensor. Check all connections, including the Sunny one on the camera board' so I am assuming I somehow managed to destroy my camera module.

roberttidey, 2019-04-19:
Before you give up on the module, check carefully the ribbon connections. They need to be fully inserted. The reference to the cable on the module is also important. That is the tiny ribbon on the module itself. Just press it down firmly.

If you are using a Pi Zero with the narrow cable then the connector on the Pi end is very fragile and the plastic is easily broken. If so then one can normally repair it by pressing the plastic together and securing with some good tape.

florisjan78, 2019-04-19:
I have 2 ribbons and have released the sunny connector and plugged it back in again. The situation remains the same. I probably screwed up while handling the raspberry. 
One thing that does blow my mind: raspimjpeg doesn't make the camera led come on, raspistill does..,.

roberttidey, 2019-04-19:
I don't think the LED is significant here. It just depends on where in the start up sequence with the camera the command is sent. It could be that raspimjpeg is detecting the camera connection is broken earlier and then not bothering to turn LED on.

The base issue is that if raspistill or raspivid don't get a camera connection then neither will raspimjpeg.

A clean OS install from scratch might be worth it to rule out the possibility the raspberry camera system software (which is used by raspistill and raspimjpeg) got broken somehow.

If you have another raspberry it might be worth trying the module + raspistill on that that to finally confirm the status of the module.

florisjan78, 2019-04-19:
I actually made an image before the upgrade. I've just tried restoring that, with the same result. It's definitely a hardware problem.

thc2018, 2019-12-02:
> {2019/12/02 16:19:56} Finished boxing /var/www/html/rpi2cam/media/vi_13497_20191202_161846.mp4 from Box Queue at pos 26
> {2019/12/02 16:19:56} Start boxing /var/www/html/rpi2cam/media/vi_13498_20191202_161919.h264 to /var/www/html/rpi2cam/media/vi_13498_20191202_161919.mp4 Queue pos 27
> {2019/12/02 16:19:56} Removed item from Box Queue
> {2019/12/02 16:19:58} Watchdog detected problem. Stopping{2019/12/02 16:19:59} SIGINT/SIGTERM received, stopping

My RPi_cam hang on. 
How can I reboot the Pi then Rpi-Cam watchdog detects the problem?

roberttidey, 2019-12-02:
If a macro called startstop.sh exists in the macros folder then it is called as the raspimjpeg process starts up and when it quits like in the case where the watchdog has shut the process down,

On start up the call to the macro had a parameter "start" and when it stops it has a parameter "stop".

I have enhanced this a bit so that the end condition better distinguishes the cause so that it will explicitly use the keyword "watchdog" rather than "stop" when the watchdog terminates the program. 

The macro can check its parameter and then take any action including rebooting the Pi.

thc2018, 2019-12-02:
Thanks for your answer, I will try to do that. The problem is very rare, occures 1 time per 3 month of usage