silvanmelchior/RPi_Cam_Web_Interface #493: What determines the amount of storage for videos?

twesleyb, 2019-04-24:
Thanks for making this tool silvanmelchior!

What determines the amount of storage allotted to the storage of videos on the web interface? I note that the www/diskusage.txt and wiki seem to indicate that this is the Pi's default media folder. Could I change this path to an external hard drive and increase the amount of storage space available?

roberttidey, 2019-04-24:
silvan did the original work on this but I (roberttidey) have been the main developer for the last 5 years..

Storage by default is in the media folder under the web folder e.g. /var/www/html/media. For standard systems without other storage this is on the SD card.

One can set up purging of files in the scheduler to either limit the total amount of space used or to guarantee a certain amount of free storage for other applications.

If one has external storage like a USB stick, USB hard drive or network storage then this can be used to hold the media files and then the space available is whatever those have.

The best way to use this is NOT to try to change the location of the normal media folder, but instead to use linux facilities to link the external storage to the normal media location. That way everything remains logically the same but the files are kept on the external device. fstab provides the mechanism to do this.

Note also that if a network location is used then it is advisable to use the boxing_path mechanism provided by the software. This means that the real time recording is done to local SD storage and then files are moved to the final destination as they are converted to mp4. This avoids any problems with any slow downs in network traffic in the recording process.

twesleyb, 2019-04-24:
Hi Robert, thank you for your great answer (and development of the software)!

Like you suggested, I was able to mount my HDD and bind the `/var/www/html/media` folder to my HDD by modifying my `/etc/fstab` file to include:
```
# Mount Seagate HDD.
UUID=xxxxxxxxxxxxxxxx /mnt/seagate ntfs-3g uid=1000,gid=1000,nofail,umask=007 0 0

# Bind /var/www/html/media to Seagate HDD.
/mnt/seagate/Media /var/www/html/media none bind 0 0
```
Insure that the drive is mounted with:
```
$ sudo mount -a
```
This does increase the size allotted for saving files--I can tell by checking the web browser. 
However, I can no longer record videos (clicking on the record video button on the web interface doesn't do anything). I can unmount the drive, with `sudo umount -a`, and now things work. I can record a video.

With the HDD mounted, and the Pi's terminal open, clicking on the record video button produces the following error in the Pi's terminal:
`mmal: mmal_port_disable: port vc.ril.video_encode:out:0(H264)(0x1306700) is not enabled`

Any ideas?

roberttidey, 2019-04-24:
Storage location must be owned by user www-data with read / write permissions

So change the ownership of your external storage location before you link it to media. Use chown command for this. Check it then has read / write permissions.

twesleyb, 2019-04-25:
Thanks for the advice, Robert. I appreciate your help.

I played around with chmod and the order of operations with chmod, mount, ect but nothing seemed to correctly change the ownership of my HDD subdirectory to www-data. Ultimately what worked for me was to modify my fstab such that the mount point is assigned to www-data (uid/gid = 33; see below). Now recording videos are saved to the HDD mount point and the amount of space is determined by this storage space! However, this is not an ideal solution because now the root user doesn't have access, but I can work on fixing this. 

```
# Mount Seagate HDD.
UUID=xxxxxxxxxxxxxxxx /mnt/seagate ntfs-3g uid=33,gid=33,nofail,umask=007 0 0

# Bind /var/www/html/media to Seagate HDD.
/mnt/seagate/Media /var/www/html/media none bind 0 0
```

closed 2019-04-25 by twesleyb