silvanmelchior/RPi_Cam_Web_Interface #494: motion / FIFO1 not working anymore - need debugging help

konstantinj, 2019-04-25:
Hi,

I've missed couple of versions and now in my docker version motion triggered events are not working anymore after the update.

In the schedule log I can see
`Executing macro /var/www/macros/motion_event.sh "1" `
when in
`motion_external = 2`
Also I can see
`Start capture requested from Pipe`
But no capture is made.
Issuing
`echo "im" > FIFO1`
does not do anything while
`echo "im" > FIFO11`
creates an image.

What else do I need to check?

my uconfig (tried several things and did not reset)
```
annotation %Y-%M-%D %h:%m:%s
anno_background 1
anno_text_size 25
brightness 60
exposure_compensation 10
hflip 0
vflip 0
width 768
quality 50
video_width 1296
video_height 972
image_quality 85
motion_detection true
vector_preview 0
motion_external 2
motion_noise 1001
motion_threshold 1
motion_initframes 3
motion_stopframes 450
motion_clip 3
```

and the schedule.json
```
{
  "Cmd_Poll": "0.03",
  "Mode_Poll": "10",
  "Management_Interval": "3600",
  "Management_Command": "",
  "PurgeVideo_Hours": "0",
  "PurgeImage_Hours": "0",
  "PurgeLapse_Hours": "0",
  "GMTOffset": "0",
  "PurgeSpace_ModeEx": "1",
  "PurgeSpace_Level": "10",
  "DawnStart_Minutes": "-180",
  "DayStart_Minutes": "0",
  "DayEnd_Minutes": "0",
  "DuskEnd_Minutes": "180",
  "Latitude": "52.00",
  "Longitude": "0.00",
  "Max_Capture": "120",
  "DayMode": "0",
  "AutoCapture_Interval": "0",
  "AutoCamera_Interval": "0",
  "Days": {
    "0": [
      "0",
      "1",
      "2",
      "3",
      "4",
      "5",
      "6"
    ],
    "1": [
      "0/",
      "1/",
      "2/",
      "3/",
      "4/",
      "5/",
      "6/"
    ],
    "2": [
      "0/",
      "1/",
      "2/",
      "3/",
      "4/",
      "5/",
      "6/"
    ],
    "3": [
      "0/",
      "1/",
      "2/",
      "3/",
      "4/",
      "5/",
      "6/"
    ],
    "4": [
      "0/",
      "1/",
      "2/",
      "3/",
      "4/",
      "5/",
      "6/"
    ],
    "6": [
      "0",
      "1",
      "2",
      "3",
      "4",
      "5",
      "6"
    ],
    "7": [
      "0",
      "1",
      "2",
      "3",
      "4",
      "5",
      "6"
    ],
    "8": [
      "0",
      "1",
      "2",
      "3",
      "4",
      "5",
      "6"
    ],
    "9": [
      "0",
      "1",
      "2",
      "3",
      "4",
      "5",
      "6"
    ],
    "10": [
      "0",
      "1",
      "2",
      "3",
      "4",
      "5",
      "6"
    ],
    "11": [
      "0",
      "1",
      "2",
      "3",
      "4",
      "5",
      "6"
    ]
  },
  "Commands_On": [
    "ca 1",
    "ca 1",
    "ca 1",
    "ca 1",
    "ca 1",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    ""
  ],
  "Commands_Off": [
    "ca 0",
    "ca 0",
    "ca 0",
    "ca 0",
    "ca 0",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    ""
  ],
  "Modes": [
    "",
    "ec 10",
    "ec 0",
    "ec 0",
    "ec 10",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    ""
  ],
  "Times": [
    "09:00",
    "10:00",
    "11:00",
    "12:00",
    "13:00",
    "14:00",
    "15:00",
    "16:00",
    "17:00",
    "18:00",
    "19:00",
    "20:00"
  ]
}
```

roberttidey, 2019-04-25:
First the FIFOs.

FIFO is the main command pipe for raspimjpeg to accept commands 

FIFO1 is a special pipe used to communicate motion detections FROM raspimjpeg back to the scheduler so that it can determine what to do when motion events occur. "1" signals a motion start, "0" signals a motion stop, "9" indicates raspimjpeg has changed settings so scheduler should restart.

FIFO11 (and FIFO12, .... FIFO20) are secondary command pipes that are equivalent to the main FIFO and can be used by other applications to avoid commands getting mixed up.

motion_external is set to 2 Settings can be
0 internal (recommended method for detection)
1 external use the older external motion process
2 monitor 

monitor turns on internal and runs motion detection scripts BUT does not issue motion events to the scheduler via FIFO1. This is intended for use where you want to run motion scripts to do various things but not cause the scheduler to do anything. You won't get the normal scheduler starting and stopping captures when in this mode.

So first of all change the motion detect mode under camera settings back to internal

konstantinj, 2019-04-26:
motion_external was 0 before. I was just experimenting a bit and wanted to check if the motion_event.sh script gets triggered. That's the case. What should I see in log when normal motion gets detected?

roberttidey, 2019-04-26:
A motion triggered recording looks like
`{2019/04/26 08:30:03} send smd 1
[2019/04/26 08:30:03] Start capture requested from Pipe
[2019/04/26 08:30:03] Send ca 1
{2019/04/26 08:30:03} Capturing started
{2019/04/26 08:30:14} send smd 0
[2019/04/26 08:30:14] Stop capture requested
[2019/04/26 08:30:14] Send ca 0
{2019/04/26 08:30:14} Capturing stopped
{2019/04/26 08:30:14} Add /var/www/media/vi_0002_20190426_083003.mp4 to Box Queue at pos 2
{2019/04/26 08:30:14} Executing macro /var/www/macros/end_vid.sh "/var/www/media/vi_0002_20190426_083003.h264" 
{2019/04/26 08:30:14} Start boxing /var/www/media/vi_0002_20190426_083003.h264 to /var/www/media/vi_0002_20190426_083003.mp4 Queue pos 2
{2019/04/26 08:30:15} Finished boxing /var/www/media/vi_0002_20190426_083003.mp4 from Box Queue at pos 2
{2019/04/26 08:30:15} Removed item from Box Queue
`
I notice that your commands are ca 1 and ca 0. I have ca 1; and ca 0; where the semicolon terminates a command. It is possible a change has meant that is required at the end of commands. I'll check that.

konstantinj, 2019-04-26:
Awesome. Looks like that was the issue!