silvanmelchior/RPi_Cam_Web_Interface #495: camera stream not working(there is no stream at all)

rajeshkrarora1, 2019-04-30:
stream not working
[image]

rajeshkrarora1, 2019-04-30:
tried ./Rpi_Cam_Control_interface_Installer.sh start------no resolve
[image]
[image]

rajeshkrarora1, 2019-04-30:
tried ./install.sh------same result
tried ./start.sh-------same result
restarted the system----again same result

roberttidey, 2019-04-30:
The recommended way to install as per wiki is to use the ./install.sh method. I don't support the older Rpi_Cam_Control_interface_Installer.sh any more and using it may prevent the ./install method working afterwards.

I recommend starting again with a clean raspbian, enabling camera interface and following wiki instructions using the ./install.sh method and using the defaults for the first install. You can always change settings afterwards.

If you still have a problem then check your camera connections and try using raspivid to make sure the camera is working OK.

keawepoo, 2019-07-16:
@rajeshkrarora1  Have you tried `sudo apt-get install motion` then reboot?