silvanmelchior/RPi_Cam_Web_Interface #497: GMT offset not in sync with Sun based schedule times

kkkkkbruce, 2019-05-20:
My issue is this, I live at a GTM offset of -5 but do not have Daylight savings time. My pi is set to the correct time and timezone and when I set the GMT offset in the RPI-Cam_Web_Interface to -5 and use the Sun bassed schedule to start and stop video recording, it starts and stops recording an hour earlier than the period times listed. For example if it says Sunrise at 6:04 it will start recording at 5:04. My work around is just to change my GMT offset in the settings to -4, but it is a bothersome chink in an otherwise very useful program.  The timestamp that shows in the videos/photos are correct, just the scheduler starts and stops an hour off.

roberttidey, 2019-05-20:
The scheduler uses a php function to determine sunrise and sunset times based on latitude and longitude and offset. This is then compared with the set up to determine what period of the scheduler one is in.

So I think what is happening is the sunrise and sunset times are getting confused by the offset.

There are two ways of entering an offset. The first is an absolute number. The second is by using the list of time zone areas as specified https://www.php.net/manual/en/timezones.php

You enter the corresponding region/city code instead of a fixed offset.

kkkkkbruce, 2019-05-20:
Hi roberttidey, I changed the offset to the region/city code of America/Guayaquil which is correct for my location. As before the timestamp in the actual video file name is correct as is the 'watermark' timestamp in the video. But I noticed that on the "Download Videos and Images" page the timestamp in the thumbnail titles for each video is still off by an hour. (Even after rebooting). So I assume that the scheduler will also still be off. I will confirm this tomorrow after watching what time it kicks on for the calculated 'sunrise' time.

kkkkkbruce, 2019-06-01:
Just an update. The scheduler works correctly now using the text region/city code for the timezone. On the Download Videos and Images page the thumbnail title still shows an + an hour from video termination.  I can close this as the scheduler works as it should now.

closed 2019-06-01 by kkkkkbruce