silvanmelchior/RPi_Cam_Web_Interface #498: Is there anyway to automate the time lapse conversion?

daliagachc, 2019-06-08:
it would be nice if we could convert the time lapse sequence to video on the fly after the movement detection is off

roberttidey, 2019-06-09:
You could achieve this by triggering a command macro to perform the conversion.

daliagachc, 2019-06-09:
Thank u @roberttidey. I was thinking about this but which one would it be? the detection macro? it does not pass the image name commands I believe. I also wanted to thank the team for the software. It's really nice

roberttidey, 2019-06-10:
If you are using the motion detect to control the time lapse then the detect macro would be the one to use. It has a parameter to tell you whether it is the start or end of detect.

Although it does not have a filename it is fairly easy to find the time lapse data. You can also create a dedicated folder for the time lapse images in the /etc/raspimjpeg settings (lapse_path) if you want to make it easier to see the set of image data to process.