silvanmelchior/RPi_Cam_Web_Interface #500: Installer Will Not Work With Buster

Trotter73, 2019-06-26:
Hi,

As the title says the installer will not work with the latest Raspbian Buster Release.

php7.0 needs to be replaced with php7.3 (Remove the option for 5 as its no longer available?) and libav-tools needs to be replaced with ffmpeg.

I have not tested but I think the command line syntax for avconv and ffmpeg should be the same, when I was researching time-lapse I found I could interchange the arguments. So maybe a just find-replace will do to correct things.

I cant really drive git, but can happily test if someone can...

jinksto, 2019-06-29:
After installing this on Jessie I did a dist-upgrade to buster.  So far everything seems to be fine.  The only issue I ran into was the /etc/passwd was changed to the following (probably by apache?):
www-data:x:33:33:www-data:/var/www:/usr/sbin/nologin
I changed it back to:
www-data:x:33:33:www-data:/var/www:/bin/bash

Everything seems to be working now.   I can't speak to your install issue.

roberttidey, 2019-06-29:
Thanks for the comments.

I have just returned from vacation and will be trying out Buster over the next few days and making any amendments as needed. I also have to be careful about just removing things like php version support as some users don't want to upgrade OS.

Trotter73, 2019-07-01:
Hi,

@jinksto
If you did an update you will probably find that you still have php7.0, not 7.3, I saw this and and a post in the forums saying the same.

@roberttidey 
True, I was stuck looking forward :) I have a couple of redundant cam setups I can trial stuff on if you need any assistance..

roberttidey, 2019-07-02:
I have made some changes to installer for Buster / php (defaults to 7.3)

I have tested on Apache. nginx might need further changes

Trotter73, 2019-07-05:
I'll have a play later, I did see feedback on the forums that the installer now no longer works on older releases....

Trotter73, 2019-07-09:
Hi,

Apache on Buster works as expected so far.

I tried Lighttpd, the install worked however, the Cam subfolder does not seem to be working as it does in Apache.

For example I want "mycam" as the sub folder...

Files are installed in /var/www/mycam however if I point my browser at http://myip.addr/mycam I get a 404 , I have to go to http://myip.addr/

roberttidey, 2019-07-09:
Not sure at the moment what is causing the lighttpd issue.

Can you try installing in mycam as before and checking that say the file loading.jpg exists in the /var/www/mycam folder? Then browse to http://myip.addr/mycam/loading.jpg should bring up a simple 'loading' image. This just tests the lighttpd is responding.

If that doesn't work it probably means the lighttpd.conf file format has changed.

Check /etc/lighttpd/lighttpd.conf

That should have a line server.document-root  = 

and this should point the path for the web files.

If it doesn't could you zip and attach your lighttpd.conf file

Trotter73, 2019-07-09:
Hi Bob,

I'm fairly certain the sub-folder was being used with respect to the local path, as I said /var/www/mycam contained all the usual files and folders, however using the sub-dir in the URL didnt work.

If I get 5 tonight I'll flash another SD and try again with lighttpd, I wanted to get something running so reverted to Apache.

roberttidey, 2019-07-09:
Having looked at the set up more closely, I think the lighttpd may behave differently to the apache.

The apache set up expects the sub-dir in the URL as the web root is left alone. This can make it easier if Apache is being used for other web apps as well. However, the lighttpd part of the install sets the web root up to include the sub-dir so that an access to the ip works without the sub-dir.

The difference is a bit confusing so I'll either clarify in the documentation or take out the sub-dir for lighttpd.

If you wanted to try out the latter case then you could try changing line 253 of install.sh to
sudo sed -i "s/^server.document-root.*/server.document-root  = \"\/var\/www\"/g" /etc/lighttpd/lighttpd.conf

before running it

Trotter73, 2019-07-09:
Sorry Bob, have I lead you up the garden path here ? I've not used the other servers before only Apache, I had assumed they would all work in the same fashion, should have checked first, I do apologize.

Ref the PHP version, can you chose 7.0/7.3 in the script ? The text in the script requester only shows 7.3. Not a lot of room but a note could be handy Deb 9=7 Deb 10=7.3 ?

roberttidey, 2019-07-09:
You should be able to enter any php version in the parameters at the start of the install. It defaults to 7.3 but you can change it to 7.0,7.1,7.2

closed 2019-08-09 by Trotter73