silvanmelchior/RPi_Cam_Web_Interface #506: System doesn't start in Buster

nunoperalta, 2019-08-18:
When upgraded to Buster, the system will not start, because of this:

root@raspberrypi:/# sudo su -c 'raspimjpeg > /dev/null &' www-data
This account is currently not available.
root@raspberrypi:/# sudo su -c "php /var/www/$rpicamdir/schedule.php > /dev/null &" www-data
This account is currently not available.
root@raspberrypi:/#

That's for the same reason as #500
But I don't think I should change /etc/passwd to remove "sbin/nologin" and put "bash" for "www-data", as that will cause security issues. (we don't want to have security issues on a system with a camera, right? ;) )

Using PHP 7.3.4.

roberttidey, 2019-08-18:
The camera install script does give www-data user a bash login. This is needed for both start up and to allow macro scripts to run.

The www-data user has very limited privileges and I don't think there is a particular security concern there. If you try to access a .sh script from the browser then it just returns the test; it doesn't execute it.

A clean install of buster followed by the camera software install does result in the system operational. A dist upgrade to buster does seem to revert the www-data bash set up and causes this issue, but a re-run of the install.sh should fix that.

nunoperalta, 2019-08-18:
Thank you very much for explaining. I trust what you say is correct. :)
Running "install.sh" (and a few other commands that I had to run to correct some of the errors with the upgrade!!) fixed the problem. Everything working great.

In the install, I had to write "7.3" as the PHP version, as otherwise with just "7" it will try to find "php7", which doesn't exist. But, "php" and "php7.3" exist in the system.

All good. Thank you very much.

roberttidey, 2019-08-18:
Good to hear that all is OK.

The old install.sh does just use the version number of php directly. The latest version of install.sh should convert a plain 7 into 7.3, although if a more explicit version is put in (like if it ever changes to 7.4) then it will try to install that.

closed 2019-08-19 by nunoperalta