silvanmelchior/RPi_Cam_Web_Interface #507: Include New AWB value “greyworld” for NOIR cameras.

Pi-DD, 2019-08-24:
Latest Raspberry Pi kernel includes new AWB value of “greyworld” specifically for NOIR cameras. It eliminates the “purple haze” effect that using AWB auto causes with NOIR camera. 

This is now present in both Stretch and Buster. 

BUT at the moment it seems that Motion (external motion detection sw) does not pass any parameter values to Pi camera software when used on Buster.

roberttidey, 2019-08-25:
These values are actually in raspimjpeg, nothing to do with Motion.

I'll check into adding this.

ebretteville, 2019-08-25:
Oh yes please, because I have the same webcam and I have this purple effect.

Pi-DD, 2019-08-25:
@ebretteville - this link, way down the thread, shows you how to set up a service that is a workaround for the purple haze problem by running a command at boot. Or you can do it by hand 

https://github.com/raspberrypi/firmware/issues/1167

You need to stop the camera and then do...
`sudo vcdbg set awb_mode 0`
Then restart camera. You can reverse this by replacing 0 with 1. 

@roberttidey - I discovered this via another setup I have that consolidates multiple cameras through a MotionEye hub. I use your fine software at a remote location with only one camera on a Pi Zero W. I use your internal motion detection algorithm, so not familiar with how Motion is used instead. 

From my other installation I learned that Motion has a parameter mmalcam_control_params through which you can pass parameters to raspimjpeg. Except that it does not seem to work on Buster installations. If your setup does not enable access to that there should be no problem for anyone using the Motion algorithm. I have posted an issue about this for Motion.

wingers999, 2019-08-25:
sudo vcdbg set awb_mode 0  - worked for me, but also setting Colour Effect to Enabled under Camera Settings in RPi Cam Web interface resolves is as that is how I did it in past

roberttidey, 2019-08-25:
I have updated the main software to latest MMAL libraries and added greyworld as an awb option.

You do need to have updated kernel / videocore as otherwise it will log an illegal value if this is selected.

Pi-DD, 2019-08-27:
Thanks Robert!
Nothing to do with your software, but FYI for those reading this...…

- Motion is not yet modified to handle "greyworld" and will default to auto if you set this. 
- failure to pass mmalcam_control_params via Motion seems to be because I had installed it with apt-get, not by downloading and then dpkg. There is a note in the documentation that installation with apt-get does not fully (at all?) configure mmal camera - even though in my case the mmal camera was shown as such.  Installed 4.2.2 by download and dpkg and params now pass.

closed 2019-08-27 by Pi-DD

JessicaWoods03, 2019-11-24:
you saved my life and my sanity- super thanks- I thought this was going to be another all day tutorial wtf moment- working so hard to get this project going- I upgraded my business turned purple and I screamed at the panda tower-