silvanmelchior/RPi_Cam_Web_Interface #508: PHP Warning:  A non-numeric value encountered in /var/www/html/schedule.php on line 518

ebretteville, 2019-08-25:
Since today (after buster upgrade and php 7.3 as I'm now on buster).
The line in the warning:

`$fMins = substr($fMins, 0, $j) * 60 + substr($fMins, $j+1);`

Thx for your work

roberttidey, 2019-08-25:
It sounds like an invalid time field exists in one of the scheduler fixed time periods. 

They should all have hh:nn in them (hours:minutes)

If you can't find this could you attach a zip of your schedule.json file from the web folder so I can investigate further.

ebretteville, 2019-08-25:
[image]
Maybe this is the problem, I can't find the schedule.json

roberttidey, 2019-08-25:
That is the folder holding the installation material. 

The run time web folder will be under /var/www  (e.g. /var/www/html)

ebretteville, 2019-08-25:
Here is the content of my schedule.json:
`{"Fifo_In":"\/var\/www\/html\/FIFO1","Fifo_Out":"\/var\/www\/html\/FIFO","Cmd_Poll":"0.03","Mode_Poll":"10","Management_Interval":"3600","Management_Command":"","PurgeVideo_Hours":"0","PurgeImage_Hours":"0","PurgeLapse_Hours":"0","GMTOffset":"1","PurgeSpace_ModeEx":"0","PurgeSpace_Level":"10","DawnStart_Minutes":"-180","DayStart_Minutes":"0","DayEnd_Minutes":"0","DuskEnd_Minutes":"180","Latitude":"47.397295","Longtitude":"0.689061","Max_Capture":"0","DayMode":"2","AutoCapture_Interval":"0","AutoCamera_Interval":"0","Days":[["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["1","2","3","4","5"],["1","2","3","4","5"]],"Commands_On":["im","","","ca 1","","","im","","","","","","","","","",""],"Commands_Off":["","","","ca 0","","","","","","","","","","","","",""],"Modes":["md 1","em night","md 1;em night","em auto","md 0;em night","md 0","md 1","","","","","","","","","",""],"Times":["08:00","06:00","","","","","","","","","",""]}`

roberttidey, 2019-08-25:
I think you are getting a warning because some of your time fields are blank.

I think this is harmless as such but I'll add something in to check for this.

roberttidey, 2019-08-26:
Latest version should handle this without warnings

ebretteville, 2019-08-30:
Maybe, I reinstalled from the groundup.

closed 2019-08-30 by ebretteville