silvanmelchior/RPi_Cam_Web_Interface #510: Pimoroni Pan-Tilt Hat support

druck13, 2019-08-27:
Would it be possible to add support for the [Pimoroni Pan-Tilt Hat](https://shop.pimoroni.com/products/pan-tilt-hat)?
Python API at https://github.com/pimoroni/pantilt-hat

roberttidey, 2019-08-27:
Currently this project supports pipan and servoblaster methods to control pan and tilt. Both these methods support a command pipe that the web software uses to send down the instructions to control pan and tilt. For example, with Pipan the web software writes to FIFO_pipan the command  "servo $pan $tilt" (where $pan and $tilt are the pan and tilt values).

The pimoroni hat uses i2c to send commands which might be a little tricky to use directly. However they have a library which gives access to the functions from python.

So a good approach would be to have a simple python program which runs, waits on a pipe for commands (which could be the existing pipan ones) on a pipe and then activates the appropriate pimoroni function. This would mean the web software would be unchanged and the pimoroni support would just be a separate python program.

I don't have a pimoroni hat to do any development but I can give you some help with this approach if you need it. If this works out I can include any python program with the distribution.

35nine, 2019-08-30:
https://learn.sparkfun.com/tutorials/setting-up-the-pi-zero-wireless-pan-tilt-camera/all

useful?

@roberttidey, might be interesting for you also?

balda, 2019-09-08:
I made a simple script `pipan_pipe.py` (after installing python libraries)

```bash
#!/usr/bin/env python

import time
import os, sys
import pantilthat

while True:
  pipein = open("/var/www/html/FIFO_pipan", 'r')
  line = pipein.readline()
  line_array = line.split(' ')
  if line_array[0] == "servo":
    pan = float(line_array[1])
    tilt = float(line_array[2])
    angle_pan = round(pan/255*180) - 90
    angle_tilt = round(tilt/255*180) - 90
    pantilthat.pan(angle_pan)
    pantilthat.tilt(angle_tilt)
  pipein.close()
```

roberttidey, 2019-09-08:
That sounds good.